prusa3d/PrusaSlicer #5051: incorrect slicing

Vozka, 2020-11-01:
### Version
2.2.0 also same problem in 2.3.0 alpha 2

### Operating system type + version
PrusaSlicer
Version:   2.3.0-alpha2+win64
Build:     PrusaSlicer-2.3.0-alpha2+win64-202010241601

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,035MB
OpenGL installation
GL version:   4.6.0 NVIDIA 451.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 960M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 298 lines cut ...]
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
In correct object (Manifold OK)
Default settings
No changes, just add object and click slice.
Generates gap in sliced object
Same result in Slicer 2.2.0 also 2.3.0alpha2
Same issue in 90°rotation but in different place and orientation
[drzak.zip](https://github.com/prusa3d/PrusaSlicer/files/5470938/drzak.zip)
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Vozka, 2020-11-01:
[image]
Original shape should be like that

neophyl, 2020-11-01:
The model has 63 non manifold edges and 225 intersecting faces according to blenders 3d print tool.  Also running the fix through netfabb option in Prusa Slicer on win10 fixes the model and lets it slice correctly.

Vozka, 2020-11-01:
oookay 
Thanks for your analysis.
I did not expectad that.
Created and generated from FreeCad.
Prusa Slicer says Manifold: Yes
So I did not expected this kind of problem. 
Thanks again.
[image]

bubnikv, 2021-09-25:
Would you please test with the latest alpha? We fixed some model import issues in PrusaSlicer 2.3.3 I believe.

closed 2026-09-29 by prusaslicer-bot[bot]