prusa3d/PrusaSlicer #5053: Linux GTK3 Perimeters box only has '-' button

fasaxc, 2020-11-01:
### Version
2.3.0-alpha.2

### Operating system type + version

Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)

Prusa MK3/s

### Behavior

On the Layers and Perimeters page, the "Perimeters" box has a `-` button next to it but there's no `+`. `-` works as expected but `+` seems like it should be there.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

fasaxc, 2020-11-07:
Seems fixed in 2.3.0-alpha3 :-)

fasaxc, 2020-11-07:
Note: tried the build that was not marked GTK3 and that works fine.

GTK3 version emits errors like this:
```
(prusa-slicer:347661): Gtk-CRITICAL **: 09:58:39.779: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
```

pmjdebruijn, 2020-11-12:
2.3.0~alpha3+69~g932dfb4fc still has this issue

YuSanka, 2020-11-13:
> `2.3.0~alpha3+69~g932dfb4fc` still has this issue
You are right. Fixed with https://github.com/prusa3d/PrusaSlicer/commit/172c1bbab7b1576b43ee2a6bdb7da8adc9736f61
Thanks.

bubnikv, 2020-11-21:
Fixed in PrusaSlicer 2.3.0-alpha4. Closing.

closed 2020-11-21 by bubnikv