prusa3d/PrusaSlicer #5054: Potential Bug in Support Generation: Pattern Spacing

Luxovious, 2020-11-01:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro 1909

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3 upgraded to mk3s 

### Behavior
* When generating support for this particular model, the support seems to go "through the walls" of the model.
* This behavior seems to intensify when the Pattern spacing is enlarged.
  * 2 mm pattern spacing does seem to work, however in this model the print time is almost one day longer
  * 2,5 mm pattern spacing seems to create small issues (see first picture attached)
  * 10 mm pattern spacing seems to create a lot of mess (see second picture attached)

### Screenshots
[image]
[image]

### Project File (.3MF) and .stl where problem occurs
[Pumpkin.zip](https://github.com/prusa3d/PrusaSlicer/files/5471427/Pumpkin.zip)

togomedia, 2021-01-02:
This issue is present in the most current MacOS version RC2 as well.

jschrempp, 2021-01-13:
Here is another example file showing the problem.

[image]

[support_through_the_roof.zip](https://github.com/prusa3d/PrusaSlicer/files/5810085/support_through_the_roof.zip)

bubnikv, 2021-02-19:
I have added filtering step, that in most cases avoids the supports to leak through the object wall.

PrusaSlicer 2.3.0, support interface spacing 10
[image]

Current master, support interface spacing 10
[image]

PrusaSlicer 2.3.0, more realistic support interface spacing 5
[image]

Current master, more realistic support interface spacing 5
[image]

Thus the support may leak only around the object and only in the cell of the interface spacing size.

commit fb7b9950 (2021-02-19): Improvements in FDM support generator:
Projection into a grid has been reworked to use the AGG rasterizer.
This fixes #5209 and #6067.
Also the raster is now being oversampled by maximum 8x8 samples
and the supports are only allowed to expand inside the cell.
This significantly reduces leakage of supports through object walls,
which fixes #5054.
closed 2021-02-19 by bubnikv