prusa3d/PrusaSlicer #5055: Display after slicing does not show fill or brim

JPS952, 2020-11-01:
### Version
Version 2.2.0 +Win64

### Operating system type + version
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,130MB
OpenGL installation
GL version:   4.6.0 NVIDIA 445.75
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 710/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Ender 5 
### Behavior
* _Describe the problem_
After slicing my spacers show no infill, just an outline, nor is the brim showing.
Both infill and brim show during the last few seconds of slicing, but once complete they disappear.
While slicing:
[image]
After slicing:
[image]

_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[8mm x 5mm spacer.zip](https://github.com/prusa3d/PrusaSlicer/files/5471546/8mm.x.5mm.spacer.zip)

neophyl, 2020-11-01:
Shows up for me when I load your project in and slice it.  Possibly stupid question but you do have those features turned on in the preview dont you ?  If you turn off skirt and gapfill then they show when slicing (gren for skirt) but once actually sliced and it colour codes everything then if they arent enabled they dont show.

[image]

[image]

JPS952, 2020-11-01:
Yes, a dumb lapse on my part. That fixed it. Thanks for straightening me out.Regards,Joe

closed 2020-11-01 by rtyr