prusa3d/PrusaSlicer #5056: Skirt Does not Account for Brim Size

ahandyman59, 2020-11-01:
### Version
_Version of PrusaSlicer used goes here_
V2.2.0+win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa I3 MK3S
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
_Is this a new feature request?_

When printing parts with a brim, the brim overlays the skirt. Shouldn't the skirt be just beyond the base of whatever is on the print bed? Picture included...
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-11-01:
Intended behavior. 

Why:
https://github.com/prusa3d/PrusaSlicer/issues/724

Duplicate issue + links to many other similar issues:
https://github.com/prusa3d/PrusaSlicer/issues/3692 

Also related:
https://github.com/prusa3d/PrusaSlicer/issues/4802

Closing.

closed 2020-11-01 by rtyr

lukasmatena, 2021-08-12:
Solved in 2.4.0-alpha1: https://github.com/prusa3d/PrusaSlicer/issues/4802#issuecomment-897492086.