prusa3d/PrusaSlicer #5057: [Bug] Arrow Keys not raising and lowering layer

mneumark, 2020-11-02:
### Version
2.3.0 alpha2+

### Operating system type + version
macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S 3.9.1

### Behavior
* In 2.2.0, after slicing, I could, after clicking once on the arrows on the right side, change the layer I was viewing by hitting the up and down arrows on the keyboard. In 2.3.0 alpha2+ I cannot.
* Slice any object. Drag on the upper arrow on the right side to change the layer. Hit the up and down arrows.
* Expected
  * Layers go up or down

JohnOCFII, 2020-11-02:
> * Slice any object. Drag on the upper arrow on the right side to change the layer. Hit the up and down arrows.

It its working for me.  Also on Catalina 10.15.7

mneumark, 2020-11-02:
I just realized what was happening. It used to allow arrow up and down after that until you clicked somewhere else. Now the arrow keys only function when the mouse is hovering over the vertical bar to the right of the sliced object(s).

NameOfTheDragon, 2020-11-04:
> I just realized what was happening. It used to allow arrow up and down after that until you clicked somewhere else. Now the arrow keys only function when the mouse is hovering over the vertical bar to the right of the sliced object(s). That still seems like a bug to me.

I can confirm the same behaviour in Windows. This is confusing, it was better when up/down arrow would work regardless of focus or mouse position.

olivbd, 2020-11-14:
Same here on ArchLinux with 2.3.0-alpha3... Arrow keys are working only if the mouse is hovering over the vertical bar.
It's a strange behaviour!

bubnikv, 2020-12-07:
This should be fixed in PrusaSlicer 2.3.0-beta2.
Closing

closed 2020-12-07 by bubnikv