prusa3d/PrusaSlicer #5058: Linux Form GIT Build Failure: WX-3.0 issue again?

clytle374, 2020-11-02:
Current git version_2.3.0-alpha2-37-g0798fa818 on Gentoo Linux.  Failure starts with commit 7c571c1d9d33150cdc60780876b275421dbdd9b8

Fails with
`[ 98%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/PrintHost.cpp.o
/home/cory/sources/prusa/src/slic3r/Utils/Repetier.cpp: In lambda function:
/home/cory/sources/prusa/src/slic3r/Utils/Repetier.cpp:206:74: error: cannot convert ‘boost::property_tree::basic_ptree<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >::data_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
  206 |                         groups.push_back(wxString::FromUTF8(v.second.data()));
      |                                                             ~~~~~~~~~~~~~^~
      |                                                                          |
      |                                                                          boost::property_tree::basic_ptree<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >::data_type {aka std::__cxx11::basic_string<char>}
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /home/cory/sources/prusa/src/slic3r/pchheader.hpp:95,
                 from /home/cory/prusaslic3r/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~
[ 98%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/Bonjour.cpp.o
[ 98%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/PresetUpdater.cpp.o
[ 98%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/Process.cpp.o
[ 99%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/UndoRedo.cpp.o
[ 99%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/HexFile.cpp.o
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:1494: src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/Repetier.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1480: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
`
Thanks, will try to provide any requested details. 
Cory

lukasmatena, 2020-11-04:
> WX-3.0 issue again?

Afraid so. Should be fixed with ab6af3ff6881395a57484351d17b54aec480d986.
Thanks. Closing.

closed 2020-11-04 by lukasmatena

clytle374, 2020-11-07:
Just a quick question as I'm now getting runtime failures.  It's looking for xwgtk in a temp folder that is used by my system during build. It's a tmpfs that only lives in memory while building the wxgtk package.  Can you provide any clues as to how it's looking there?  Not sure what error it could be in my system since everything thing else wxgtk works on my system.

Thanks
`/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/gtk/dc.cpp(298): assert "cr" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

(prusa-slicer:27236): Gtk-WARNING **: 06:56:34.103: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(423): assert "IsOk()" failed in SetTextForeground(): wxGCDC(cg)::SetTextForeground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:35.281: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(437): assert "IsOk()" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:36.050: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(1160): assert "m_graphicContext" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:36.829: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/gtk/dc.cpp(298): assert "cr" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

(prusa-slicer:27236): Gtk-WARNING **: 06:56:37.866: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(423): assert "IsOk()" failed in SetTextForeground(): wxGCDC(cg)::SetTextForeground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:39.047: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(437): assert "IsOk()" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:39.967: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(1160): assert "m_graphicContext" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:40.714: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/gtk/dc.cpp(298): assert "cr" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

(prusa-slicer:27236): Gtk-WARNING **: 06:56:41.699: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(423): assert "IsOk()" failed in SetTextForeground(): wxGCDC(cg)::SetTextForeground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:42.685: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(437): assert "IsOk()" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:43.587: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(1160): assert "m_graphicContext" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:44.735: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/gtk/dc.cpp(298): assert "cr" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

(prusa-slicer:27236): Gtk-WARNING **: 06:56:46.871: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(423): assert "IsOk()" failed in SetTextForeground(): wxGCDC(cg)::SetTextForeground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:47.871: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(437): assert "IsOk()" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:48.454: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(1160): assert "m_graphicContext" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:49.239: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/gtk/dc.cpp(298): assert "cr" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

(prusa-slicer:27236): Gtk-WARNING **: 06:56:50.189: gtk_window_set_titlebar() called on a realized window
[... cut ...]
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(437): assert "IsOk()" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:51.738: gtk_window_set_titlebar() called on a realized window
/var/tmp/portage/portage/x11-libs/wxGTK-3.0.4-r302/work/wxWidgets-3.0.4/src/common/dcgraph.cpp(1160): assert "m_graphicContext" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

(prusa-slicer:27236): Gtk-WARNING **: 06:56:52.325: gtk_window_set_titlebar() called on a realized window
Trace/breakpoint trap
`