prusa3d/PrusaSlicer #5062: Custom Controls related traces

pmjdebruijn, 2020-11-02:
### Version
`2.3.0~alpha2+94~g186d408cf`

### Operating system type + version
xubuntu 20.04 LTS with wxGTK 3.1.4

### Behavior
When switch to the print settings tabs, I get these:

```
ASSERT INFO:
../src/gtk/dc.cpp(343): assert ""cr"" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

BACKTRACE:
[1] wxNativeDCFactory::CreatePaintDC(wxPaintDC*, wxWindow*)
[2] wxPaintDC::wxPaintDC(wxWindow*)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxNotebookBase::SendPageChangedEvent(int, int)
[9] g_closure_invoke
[10] g_signal_emit_valist
[11] g_signal_emit
[12] g_closure_invoke
[13] g_signal_emit_valist
[14] g_signal_emit
[15] gtk_main_do_event
[16] g_main_context_dispatch
[17] g_main_loop_run
[18] gtk_main
[19] wxGUIEventLoop::DoRun()
[20] wxEventLoopBase::Run()
[21] wxAppConsoleBase::MainLoop()
[22] wxEntry(int&, wchar_t**)
[23] __libc_start_main
```

```
ASSERT INFO:
../src/common/dcgraph.cpp(431): assert ""IsOk()"" failed in SetTextForeground(): wxGCDC(cg)::SetTextForeground - invalid DC

BACKTRACE:
[1] wxGCDCImpl::SetTextForeground(wxColour const&)
[2] wxDCImpl::InheritAttributes(wxWindow*)
[3] wxNativeDCFactory::CreatePaintDC(wxPaintDC*, wxWindow*)
[4] wxPaintDC::wxPaintDC(wxWindow*)
[5] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[6] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[7] wxEvtHandler::TryHereOnly(wxEvent&)
[8] wxEvtHandler::ProcessEventLocally(wxEvent&)
[9] wxEvtHandler::ProcessEvent(wxEvent&)
[10] wxNotebookBase::SendPageChangedEvent(int, int)
[11] g_closure_invoke
[12] g_signal_emit_valist
[13] g_signal_emit
[14] g_closure_invoke
[15] g_signal_emit_valist
[16] g_signal_emit
[17] gtk_main_do_event
[18] g_main_context_dispatch
[19] g_main_loop_run
[20] gtk_main
[21] wxGUIEventLoop::DoRun()
[22] wxEventLoopBase::Run()
[23] wxAppConsoleBase::MainLoop()
[24] wxEntry(int&, wchar_t**)
[25] __libc_start_main
```

```
ASSERT INFO:
../src/common/dcgraph.cpp(445): assert ""IsOk()"" failed in SetTextBackground(): wxGCDC(cg)::SetTextBackground - invalid DC

BACKTRACE:
[1] wxGCDCImpl::SetTextBackground(wxColour const&)
[2] wxDCImpl::InheritAttributes(wxWindow*)
[3] wxNativeDCFactory::CreatePaintDC(wxPaintDC*, wxWindow*)
[4] wxPaintDC::wxPaintDC(wxWindow*)
[5] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[6] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[7] wxEvtHandler::TryHereOnly(wxEvent&)
[8] wxEvtHandler::ProcessEventLocally(wxEvent&)
[9] wxEvtHandler::ProcessEvent(wxEvent&)
[10] wxNotebookBase::SendPageChangedEvent(int, int)
[11] g_closure_invoke
[12] g_signal_emit_valist
[13] g_signal_emit
[14] g_closure_invoke
[15] g_signal_emit_valist
[16] g_signal_emit
[17] gtk_main_do_event
[18] g_main_context_dispatch
[19] g_main_loop_run
[20] gtk_main
[21] wxGUIEventLoop::DoRun()
[22] wxEventLoopBase::Run()
[23] wxAppConsoleBase::MainLoop()
[24] wxEntry(int&, wchar_t**)
[25] __libc_start_main
```

```
ASSERT INFO:
../src/common/dcgraph.cpp(1232): assert ""m_graphicContext"" failed in DoGetTextExtent(): wxGCDC(cg)::DoGetTextExtent - invalid DC

BACKTRACE:
[1] wxGCDCImpl::DoGetTextExtent(wxString const&, int*, int*, int*, int*, wxFont const*) const
[2] wxDCImpl::GetMultiLineTextExtent(wxString const&, int*, int*, int*, wxFont const*) const
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxNotebookBase::SendPageChangedEvent(int, int)
[9] g_closure_invoke
[10] g_signal_emit_valist
[11] g_signal_emit
[12] g_closure_invoke
[13] g_signal_emit_valist
[14] g_signal_emit
[15] gtk_main_do_event
[16] g_main_context_dispatch
[17] g_main_loop_run
[18] gtk_main
[19] wxGUIEventLoop::DoRun()
[20] wxEventLoopBase::Run()
[21] wxAppConsoleBase::MainLoop()
[22] wxEntry(int&, wchar_t**)
[23] __libc_start_main
```

After clicking through a whole bunch of those, the UI looks like it's had better days:
[image]

pmjdebruijn, 2020-11-04:
@YuSanka: this still applies to current git master 2.3.0~alpha3+1~gea5fdcd7b

pmjdebruijn, 2020-11-11:
@YuSanka it seems a2b915af0fafd7a52758dcef6f45647953ff3bd5 may have resolved this, I'll do some more testing over the coming days.

pmjdebruijn, 2020-11-11:
BTW, the background color difference visible in my screenshot is still an issue though

YuSanka, 2020-11-12:
Thanks for your reports. Fix of this bug will be a part of the upcoming release

closed 2020-11-12 by YuSanka