prusa3d/PrusaSlicer #5063: Layer Shifting

heitzer95, 2020-11-02:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mini and FIRMWARE 4.2.1

### Behavior
I currently have random layer shifting. I already did the recommendations in the manual but the problem is still there. The problem is reproducable if I print the gcode again. It happens on exakt the same layers.
[image]
[image]

 If I slice the part again sometimes the layer shifting is gone...

#### Project File (.3MF) where problem occurs

neophyl, 2020-11-02:
Layer shifting is almost always down to physical problems.  Although there have been occasional problems when using octoprint with some of the plugins, but that's rare from what I have seen.
Personally those shifts look like what I would expect with a loose drive pulley on one of the stepper motors.  As they aren't on your build plate I have no reference as to the direction they were placed so you should check both X and Y stepper pulleys to make sure the set screws are still tight.  When the pulleys looses they allow a bit of play which results in small shifts back and forth like pictured.  As they get looser and looser the amount of play increases leading to bigger and bigger shifts.

Its worth checking and if they are loose don't forget to re-tighten them in the correct order.

heitzer95, 2020-11-02:
Thanks for the reply. 👍 

It's the x-axis and I checked the belt tension as said in the manual. As the layers where the shifts happen are exactly the same on both prints and that the problem not happend after I sliced the part again, I though it wouldn't be the pulley. But so I have to check it again and will post if it was the troublemaker.

bubnikv, 2020-12-07:
I am closing this issue, most likely it is a HW problem.

closed 2020-12-07 by bubnikv