prusa3d/PrusaSlicer #5064: Drag and Drp .3mf Naming

GWdd, 2020-11-03:
### Version
2.3.0.a2

### Operating system type + version
_Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
Mk3

### Behavior

Load a 3mf project.  Slicer asserts that name to the task bar and subsequent gcode exports. All good.

Now, delete that part, and drag a NEW 3mf PROJECT to the platter.  The new 3mf populates as expected.  Project name in the title bar, and subsequent saves uses the OLD project, not the new project.

I have the expectation that bringing in a new project, either by FILE/OPEN_PROJECT or DRAG-DROP should behave the same. 

The failure here is that if you drag in a new project, edit it then save it, you over-write something that has no association with current work.

#### Project File (.3MF) where problem occurs
And two projects will show this behavior.

bubnikv, 2020-11-24:
Fixed with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha4

closed 2020-11-24 by bubnikv

GWdd, 2020-11-24:
...

GWdd, 2020-11-24:
...