prusa3d/PrusaSlicer #5067: [2.3.0-alpha2] Generating g-code lasts very very long (infinite loop?)

eurohell, 2020-11-03:
### Version
2.3.0-alpha2

### Operating system type + version
Windows 10 Pro 10.0.19041

### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S, 3.9.1

### Behavior
Slicing attached file with the new seam placing by painting function and forbidding seams on the outer side of the "ropes" of the model lasts infinite.

Slicing only one of the objects works flawless, up to 2 rows of 7 objects is possible, but lasts a few minutes and slicing all 12 rows of 7 objects doesn't finish after several hours of slicing.

#### Project File (.3MF) where problem occurs
[Kalkstein_12x5er_TEST_Kastanienbraun3_Weiss4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5482634/Kalkstein_12x5er_TEST_Kastanienbraun3_Weiss4.3mf.zip)

lukasmatena, 2020-11-04:
It seems to hang inside `Slic3r::offset_ex` function. It works when there is only one object and all your objects are identical, I don't know what's different when there are more. I'll add this to my list.

You should not do it like this btw - if you want to print the same object 84 times, add just one object and set the number if instances to 84 (right click on object->Set number of instances). This will make the job much easier for slicer, it will only slice the object once. And even the G-Code export will finish.

NameOfTheDragon, 2020-11-04:
I've had this happen when there is only a single object, so I'm not convinced it is due to multiple objects or instances. My "gut instinct" is that it seems to be related to the solid infill type, the new monotonic infills seem to cause a noticeable slowdown, but I can't prove it at present.

lukasmatena, 2020-11-04:
The original problem reported by @eurohell is definitely tied to some horrible naive "algorithms" when applying custom seam. These are known to be there and I intended to replace them with better ones. I didn't realize it will be this bad though. This is not fixed in 2.3.0-alpha3 (which was just released), but it will be in the next one, it is now a priority.

@NameOfTheDragon If you see it freeze when no custom seam is applied, it would be best to file a separate issue and post a 3MF so we can try to reproduce it. It is quite possible that there is another place where bad things happen.

lukasmatena, 2020-11-27:
Should be improved with 22f93a3 for PrusaSlicer 2.3.0-beta1. The 3MF slices in a reasonable time now. Closing.

@eurohell Again, you should absolutely use instances for things like that. It will slice much faster (custom seam or not).

closed 2020-11-27 by lukasmatena