prusa3d/PrusaSlicer #5068: [2.3.0-alpha2 - feature request]: Automatically surf to URL for printer host (if desired)

Boby71, 2020-11-03:
This is a feature requets:
If you add this option here, you could give the user the chance to start the default webbrowser with the given IP adress and bring him directly to the OctoPrint interface after the print job has been sent to Octoprint.
If the user don't want the browser to open up automatically, he just need to deacitvate the checkbox. 
[image]

bubnikv, 2020-12-07:
Our SLA guys were asking for this feature. I am a bit worried about what URL should PrusaSlicer open. Indeed, various print hosts support weird URL formats, for example adding HTTP basic authentization or paths or whatever we may be yet surprised with.

Boby71, 2020-12-07:
I’d say just open the defined URL which is being set for the Octoprint server. Means: Just open the default browser and use the set URL. The authentication inside Octoprint should not worry PrusaSlicer at all: Your job is done as soon as the browser has been launched.

bubnikv, 2020-12-07:
For example, here one enters the user name and password as part of the URL.
https://github.com/prusa3d/PrusaSlicer/issues/650

Boby71, 2020-12-10:
My simple view: Authentication in Octoprint should not be a job of PrusaSlicer. There are ways to e.g. suppress the authentication in Octoprint at all or to sign automatically in, if you connect from the local network. My request was only intended to simplify the workflow when launching a print: Today I need to a) export the file and b) start the browser and c) open the Octoprint server website. 

a), b) and c) could be reduced to a) alone.

I, for example, am continuously logged in in Octoprint: When I open my internal url, I'm signed in already without the need to enter credential. But that's something I have to achieve inside Octoprint, not in PrusaSlicer. That's just my point of view.

closed 2026-09-29 by prusaslicer-bot[bot]