prusa3d/PrusaSlicer #5071: Reload from disk for .3mf is unavailable

SigmaRelief, 2020-11-04:
### Version
2.3.0 Alpha 2

### Operating system type + version
Windows 10 Pro 64 bit 2004

### 3D printer brand / version + firmware version (if known)
MK3S, lightly modified 3.9.X firmware

### Behavior
Summary - Right click on object "Reload from disk" is greyed out for .3mf file even when "Export sources full path name..." is selected in Preferences.

Details
-Launch Prusa Slicer
-Ensure Preferences option "Export sources full pathname..." is selected
-Import any .3mf fie
-Optionally Save project (does not change result)
-Right click on object, "Reload from disk" is unavailable

test .3mf and .stl files were made with SOLIDWORKS 2019, stl functions as expected while .3mf does not.

Version 2.2.0 Alpha 2 release notes indicated that reload was expanded to .3mf and .amf, I vaguely remember it working on some of the early 2.2.0 betas, but it has never worked for me on multiple PCs for the final release version of 2.2.0.

#### Project File (.3MF) where problem occurs
[image]
[image]
[Project and source files.zip](https://github.com/prusa3d/PrusaSlicer/files/5485172/Project.and.source.files.zip)

Hooch180, 2020-11-04:
Exactly the same issue for me. Only difference, I'm using SolidWorks 2020.
Please take a look at it.

SigmaRelief, 2020-12-03:
I just confirmed this is still unavailable in 2.3.0 Beta 1.  Please consider checking this as it was functional for a brief moment in an earlier release series.

SigmaRelief, 2020-12-21:
It looks like reload for .amf was fixed in rc 1  based on the notes ('Reload from disk' did not work for AMF files (#5443)), but I confirmed reload for .3mf is still unavailable.  Considering the underlying similarities in the file types and that .3mf is the prefered modern alternative, the .amf fix may be adaptable to .3mf.

rtyr, 2021-06-09:
Fixed on master. The fix will be included in PS2.4.0-alpha1. Closing.

closed 2021-06-09 by rtyr