prusa3d/PrusaSlicer #5073: Gaps in Corners of Large parts

Martin-Lotz, 2020-11-04:
### Version
2.2 and 2.3 alpha 2

### Operating system type + version
Windows 10, 64 bit

### 3D printer brand / version + firmware version (if known)
Cr-10 S5, Mostly Stock

### Behavior
First up, if this has already been solved else ware feel free to point me in the right direction 😉

Hi all, got a little something that needs looking into. For work I have been printing some large prototype enclosures using the most of the S5's build Volume. When slicing the lower half there have been some strange Gaps forming at the corners of the part. 

**Model**
[image]

**Issue**
[image]
[image]

I have had the same behaviour in both the 2.2.0 and 2.3.0-alpha_2 versions.

It seems to me to be caused by the size of the model... When scaling the part down these Gaps go away _(Kind of)._ Note that the model is natively that big and is not being scaled up.

**Model Info**
[image]

**Scaled down to 50%**
[image]

Its strange because the path generation seems to work fine, and only after the feature recognition etc. do the gaps show up. 
[image]

Could this be a limitation somewhere under the hood? Thanks in advance.

Also:
- I tried slicing in Cura and it seemed to work (But I would like to stay using PrusaSlicer if i can 😉)
- The model dosen't appear to have any issues, having a look at it in Meshmixer
- Same issue no matter what detail I use to export the mesh from Fusion 360

rtyr, 2020-11-04:
Can you provide some data? 3MF, gcode or at least your exported PrusaSlicer settings.

Martin-Lotz, 2020-11-04:
Yip sorry about that. Here you go. Had to get the green light to share the file.
[Main_Body.zip](https://github.com/prusa3d/PrusaSlicer/files/5488371/Main_Body.zip)

rtyr, 2020-11-04:
It is what I suspected. 

Add `G92 E0` command to "before layer change g-code", it will fix the issue.

See this issue for further explanation https://github.com/prusa3d/PrusaSlicer/issues/4881. 

[image]

Martin-Lotz, 2020-11-04:
Sweet. That fixed it. 
Nothing new under the sun I guess. Thanks for the help!

closed 2020-11-04 by Martin-Lotz

rtyr, 2020-11-04:
You can remove the zip file now if you want.