prusa3d/PrusaSlicer #5074: Alpha 2.3.0 Fails to start with "Internal error: calling post_init() while not yet initialized"

JustgeekDE, 2020-11-04:
### Version
2.3.0-alpha2 Linux x64
2.3.0-alpha2 Linux x64 GTK3

### Operating system type + version
Ubuntu 18.04.4 LTS
`Linux xps13 4.15.0-122-generic #124-Ubuntu SMP Thu Oct 15 13:03:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s + MMU2s

### Behavior
On the first start of the Prusaslicer 2.3 Alpha everything worked correctly. Subsequent starts fail with the error pop up:

`Internal error: calling post_init() while not yet initialized`

While trying out the Alpha 2.3.0 I have also continued to use the version 2.2.0, in case that might have caused problems.
I've also attached the error.log from the console.

[error.log](https://github.com/prusa3d/PrusaSlicer/files/5487389/error.log)

bubnikv, 2020-12-07:
Do you have this issue with PrusaSlicer 2.3.0-beta2?

lukasmatena, 2021-01-04:
Should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2). Closing.

closed 2021-01-04 by lukasmatena