prusa3d/PrusaSlicer #5075: Square Bracket in Start GCODE comment causes Slice failure

JAYTEEAU, 2020-11-04:
Previously fixed??? #4665 by phazei was closed on Sep 1

### Version
; generated by PrusaSlicer 2.3.0-alpha2+ on 2020-11-04 at 12:09:36 UTC

### Operating system type + version
Mac OS 10.10.5

### 3D printer brand / version + firmware version (if known)
Original CR-10, no mods

### Behavior
* inserting an opening square bracket [ in the comments of start gcode causes slice error
* even having a closing ] does not solve the issue. Looks like a reserved character
* the closing square bracket ] by itself does NOT cause the issue

### Log snippet - look at last couple of lines
; generated by PrusaSlicer 2.3.0-alpha2+ on 2020-11-04 at 12:09:36 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M190 S65 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature
;TYPE:Custom

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 1. Expecting tag literal-char
; JAYTEE says [ will fail ]
                     ^
!!!!! End of an error report for the custom G-code template start_gcode
[image]

neophyl, 2020-11-04:
They closed that previous issue as it was not judged as a bug.  You would have to convince them that it is a bug and change their past decision.  I cant see that any of the reasoning has changed though.  
Placeholders can be used in comments with valid reasons and by using the [] you are declaring the contents a placeholder so of course the parsing fails.  It doesn't matter that its a comment line.

JAYTEEAU, 2020-11-04:
Agreed, but if I were to have only an opening [ it still fails

Perhaps a warning in the UI or some sort of notification / "compile error" upon gcode submission

closed 2020-11-04 by JAYTEEAU