prusa3d/PrusaSlicer #5076: Export tool path as obj does not work on CLI

FlorianJa, 2020-11-04:
### Version
2.3.0-alpha2 win 64
and 2.1.1. win 64

### Operating system type + version
Windows 10 64 bit

### Behavior
The CLI command --export-obj does not export the object as aspected. There is only an empty file and no error message in the console. Related issue: #3046

To test this run 
`.\prusa-slicer-console.exe --export-obj --loglevel 5 FILENAME.stl`
or
`.\prusa-slicer-console.exe -s --export-obj --loglevel 5 FILENAME.stl`

Exporting the toolpath as obj works well with the GUI but not with the CLI

Output of the first command:
```
[2020-11-04 15:47:44.707777] [0x00005cbc] [debug]   TriangleMesh::repair() started
[2020-11-04 15:47:44.792548] [0x00005cbc] [debug]   TriangleMesh::repair() finished
[2020-11-04 15:47:44.792548] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-11-04 15:47:44.796537] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-11-04 15:47:44.804516] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-11-04 15:47:45.039887] [0x00005cbc] [debug]   TriangleMesh::repair() started
[2020-11-04 15:47:45.058837] [0x00005cbc] [debug]   TriangleMesh::repair() finished
[2020-11-04 15:47:45.058837] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-11-04 15:47:45.058837] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-11-04 15:47:45.061871] [0x00005cbc] [trace]   TriangleMeshSlicer::require_shared_vertices - end
File exported to .\FILENAME.obj
```

bubnikv, 2020-12-07:
--export-obj is to convert STL to obj.
Exporting G-code paths  never worked from command line as it uses the functionality provided by the GUI module, which is not loaded by the command line slicer.

FlorianJa, 2020-12-07:
Thanks for the feedback. That was not clear to me. But the conversion from stl to obj does not work neither.

closed 2026-09-29 by prusaslicer-bot[bot]