prusa3d/PrusaSlicer #5078: Please add “Don't Connect” Infill option as in Slic3r/SuperSlicer

togomedia, 2020-11-04:
### Version
2.3.0 Alpha 2+

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
 Model Name:	MacBook Pro
  Model Identifier:	MacBookPro16,1
  Processor Name:	8-Core Intel Core i9
  Processor Speed:	2.4 GHz
  Number of Processors:	1
  Total Number of Cores:	8
  L2 Cache (per Core):	256 KB
  L3 Cache:	16 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB

 System Version:	macOS 10.15.7 (19H2)
 
[image]
Kernel Version:	Darwin 19.6.0

### 3D printer brand / version + firmware version (if known)
MK-3 FW 3.9.1

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

[image]

I apologize for the wrong format of this request.

Please add “Don't Connect” Infill option (see attached image) as seen here in Slic3r/SuperSlicer.

This is a very useful feature and can significantly increase the visual quality of outer perimeters especially when printing objects with thin walls.

[image]

bubnikv, 2020-11-04:
What is your motivation for that?

togomedia, 2020-11-09:
This is a very useful feature and can significantly increase the visual quality of outer perimeters especially when printing objects with thin walls. Additionally, it allows for both dense and open infills to be used for visible walls where no parameter is required.

bubnikv, 2020-11-20:
PrusaSlicer 2.3.0-alpha4 will implement configurable infill anchor length. You may set it to zero.

Closing.

closed 2020-11-20 by bubnikv