prusa3d/PrusaSlicer #5079: GCode Viewer changes files modified date in windows

meisterfuu, 2020-11-04:
### PrusaSlicer-2.3.0-alpha3+win64-202011041552

### Windows 10 Pro

### Behavior
Every time i open a file with the standalone gcode viewer the file modified time gets updated. It doesn't matter if i open it by clicking on the file or opening it through the viewer ui.

This should not happen since it is only a viewer. I don't know if it behaves liek that on other OSs.

bubnikv, 2020-11-04:
@enricoturri1966 would you please verify that we open a file read only?
I believe windows track a read only access as well though.

Hooch180, 2020-11-04:
@bubnikv 
Hi,
Firstly, Windows has separate file property "Last access time".
Secondly, Windows doesn't update accessed time. It is for "performance" reasons.

meisterfuu, 2020-11-04:
I checked the creation date. It was updated as well.

bubnikv, 2020-11-20:
It will be fixed in  PrusaSlicer 2.3.0-alpha4
Closing.

closed 2020-11-20 by bubnikv

bubnikv, 2021-09-27:
Fixed with 07e7e115901c80f282b06ea6b86bc56b28e1a02b

    This is a regression wrt. PrusaSlicer 2.4.0-alpha1 due to the G-code
    processing optimization and parallelization.
    Related to GCode Viewer changes files modified date in windows #5079