prusa3d/PrusaSlicer #5080: PrusaSlicer UI issue after adding pause

JohnOCFII, 2020-11-04:
### Version
2.3.0-alpha 3+

### Operating system type + version
macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa MK3S (issue not printer specific)

### Behavior

After adding a pause, plater view changes to ColorPrint mode.  After manually changing back to <default> mode, and then changing a print setting (for example, from 3 perimeters to 4), and switching back to the Plater view, the display changes once again to show the ColorPrint view.

Steps to reproduce:
1. Put item on plater
2. Slice item
3. Use slider to choose layer to insert pause
4. Right-Click and selection option to “Add pause print (M601)”
5. Re-slice object
6. Observe plater view is now set to “Color Print”
7. Switch View to “Feature type”
8. Switch to “Print Settings”
9. Change any field (for example: Perimeters from 3 to 4)
10. Switch back to Plater
11. Re-slice
12. Note that View is set back to “Color Print”

_Is this a new feature request?_
No
[image]
[Thumb_Screw_View_IssueExample-3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5489534/Thumb_Screw_View_IssueExample-3mf.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

vranisimo, 2020-11-06:
I can confirm that on a windows OS.
Basically, it happens every time after re-slicing, it is not needed to add pause to reproduce the issue.

bubnikv, 2020-12-05:
Fixed in PrusaSlicer 2.3.0-beta2.

closed 2020-12-05 by bubnikv

JohnOCFII, 2020-12-05:
Confirmed fixed.  Works as expected now.  Thanks!