prusa3d/PrusaSlicer #5082: When "Iron All Top Surfaces" is enabled, the skirt is ironed

NameOfTheDragon, 2020-11-04:
### Version
2.3.0-alpha.1

### Operating system type + version
Windows 10 Pro x64

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 with Klipper firmware

### Behavior

When "Iron All Top Surfaces" is enabled, the second skirt layer is printed extremely slowly which wastes a LOT of time. Presumably, it is being ironed. While that is perhaps pedantically correct, the skirt should probably never be ironed, it's probably not what the user wants. There's no point in ironing "helper" extrusions.

Steps to reproduce:
Print any model with the following settings enabled:
- Infill => Enable Ironing
- Infill => Ironing Type => All Top Surfaces

#### Project File (.3MF) where problem occurs

The problem occurs with this project file. [TPL Vacuum Seal Bag Nozzle.zip](https://github.com/prusa3d/PrusaSlicer/files/5490031/TPL.Vacuum.Seal.Bag.Nozzle.zip)

The problem can be seen in the generated gcode just after the beginning of layer 2:
[image]

As you can see, the feed rate is set to F159.118 (about 2.65 mm/s) which is crazy slow. It's even way slower than the default 15% setting for ironing speed (based on a default print speed of 100 mm/s).

When ironing is disabled, the following correct gcode is generated for the skirt:
[image]

NameOfTheDragon, 2020-12-25:
This problem is still present in RC2! Ironing - one of the flagship new features - is completely and utterly broken. Why aren't you fixing this?

lukasmatena, 2020-12-25:
Thanks for bumping this. We probably missed this issue in the flood of the others.

I looked into it and the "Ironing is completely and utterly broken" is a bit exaggerated. In fact, it is **autospeed** that is broken by turning ironing on. Ironing produces very thin fill extrusions and that fools the autospeed, which then sets very small speeds. The ironing extrusions should probably be ignored for autospeed calculations. It is happening in `autospeed_volumetric_limit` function. It breaks not only the skirt, but all speed calculations. We will see how easy that'll be to fix.

By the way, what is your experience with the autospeed, if you use it regularly? I personally regret that it ever made it to slicer.

NameOfTheDragon, 2021-01-01:
@lukasmatena Good detective work. Interesting comment re: auto-speed - I rather like the auto-speed feature. I tend to control things via the `Max volumetric speed` setting. That allows me to easily vary my speeds per filament without having to have a different profile.

lukasmatena, 2021-01-04:
Should be fixed in `master` by https://github.com/prusa3d/PrusaSlicer/commit/30c8592b10954f09e78185fd06448cf7ce6064cd for 2.3.0 final release. Closing.

closed 2021-01-04 by lukasmatena