prusa3d/PrusaSlicer #5083: [2.3.0-alpha3] GCode upload to repetier server successful but file cannot be found on the server

mrcg, 2020-11-04:
### Version
Version: 2.3.0-alpha3+win64

### Operating system type + version
Windows 10 Professional 64bit

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
New Feature: Upload GCode to Repetier Server

Problem Description:
All settings for repetier server have been done (IP, API).
"TEST" says => Connection to Repetier works correctly.
After pressing the button to upload GCode to Repetier Server the status in the print host upload queue says "Completed".
However the uploaded GCode cannot be found on repetier server.
The uploaded GCode could not be found on repetier server, not in the "all" group nor any other existing group.
Refresh in browser and reboot of repetier server does also not solve the issue.

lukasmatena, 2020-11-04:
@docbobo Sorry for tagging you like this, but do you see what could the problem be upfront? I know that your PR was significantly altered before we merged it, but we tested and it seemed fine. Did you get to test PS after the merge? Could you please test 2.3.0-alpha3 if you happen to have time? Thanks.

OlaStensby, 2020-11-05:
I have the exat same experience with Alpha 3 but with Octoprint instead of repetier. Computer running on MAC OSX.

ropaha, 2020-11-05:
Same here. I can add the repetier-server IP and test the connection. While pressing "Refresh Printers" Alpha 3 crash.

userosos, 2020-11-05:
I has crash if i press to Refresh Printers after input IP and API key.

wabbitguy, 2020-11-06:
I tested Alpha 3 with OS X Mojave and Repetier Server Pro...when I select Repetier in the drop down menu, the Browse button dims, but I can put in the LAN URL and the API key and test the connection.

Once I have tested the connection, I then "refresh" the printer list (I have 3 printers). Nothing is populated by default but the three printers show up in the list and I can select one.

From that point, I can send GCODE to printer and it shows up in the default group and works just fine. I LOVE this feature!

bubnikv, 2020-11-06:
@userosos commented 23 hours ago • 
> I has crash if i press to Refresh Printers after input IP and API key.

What do you mean with "it has crash?" What exactly happens?

bubnikv, 2020-11-06:
> @ropaha commented yesterday
> Same here. I can add the repetier-server IP and test the connection. While pressing "Refresh Printers" Alpha 3 crash.

How does Alpha 3 crash? What version of Repetier server are you running?

bubnikv, 2020-11-06:
@mrcg 

What version of Repetier server are you running? Are you running the latest Repetier server?

mrcg, 2020-11-06:
I am using the latest version of repetier server (0.94.3).
I have also tried to run Prusa slicer with and without admin privilegs ... does not solve.
I have tried with very simple and short filenames (e.g. test.gcode) to assure problem does not come from long filenames or special characters ... does not solve.
I have tried with and without forward slash in front of the filename ... does not solve.
BTW ... is it necessary to put froward slash in front of the filename if no group folder is used?

The strange thing is that all uploads are showing "Completed" message in the Print Host Upload Queue window. 
When pressing the "Refresh Printers" button the software is crashing (with correct API key) as also reported from other users above. In case of wrong API key i get the message "The requested feature requires a user session with adequate permissions" which is somehow expected.

Any special logfile i could provide to deeper analyze the topic?

ropaha, 2020-11-06:
> @userosos commented 23 hours ago •
> 
> > I has crash if i press to Refresh Printers after input IP and API key.
> 
> What do you mean with "it has crash?" What exactly happens?

After a few seconds the Alpha 3 closes with no error message.

ropaha, 2020-11-06:
> > @ropaha commented yesterday
> > Same here. I can add the repetier-server IP and test the connection. While pressing "Refresh Printers" Alpha 3 crash.
> 
> How does Alpha 3 crash? What version of Repetier server are you running?

After a few seconds the Alpha 3 (Windows 10 Pro 2004) closes with no error message. I'am using Repetier-Server Pro 0.94.3 on a Raspberry Pi 3.

bartwe, 2020-11-06:
For me also, prusaslicer just goes away(crashes?) if i press 'Refresh Printers' and wait about 3 seconds when pointed at my repetier server.
Skipping that step, it says it successfully uploaded in prusaslicer, but file is not visible in repetierserver, in server.log i find:
2020-11-06 12:07:02: Websocket opened
2020-11-06 12:08:03: error accessing /printer/model/: Printer offline or usage error
it looks like /printer/model/ are default names that do not match the correct names, in my case i was expecting /printer/Sapphire2/

userosos, 2020-11-06:
> What do you mean with "it has crash?" What exactly happens?

I mean that PS was closed without an messages.

wabbitguy, 2020-11-06:
I'm using a Raspberry PI 4 but had a number of problems with Repetier Server Pro's latest version so I rolled mine back to 0.94.2.

insomniux, 2020-11-13:
Both issues mentioned in this topic occurs also here:
1. PrusaSlicer crashes when clicking Update Printer.
2. No files are uploaded when after entering the host name and api code

**However** when you insert "/printer/model/PRINTERNAME" in the field for the hostname, like 
`https://repetierserver:portnum/printer/model/PRINTERNAME `
the gcode file is uploaded.

There is another problem. It does not only upload the selected gcode under the name you enter in the dialog, but it also uploads a file which appeared to be a cachefile (eg .PrusaSlicer.upload.d46c-b4b0-f8c9-5bf4)

My Prusaslicer (alpha4) runs on debian 10, my repetierserver (0.94.3) runs on raspberry

Dalton0190, 2020-12-01:
> After a few seconds the Alpha 3 (Windows 10 Pro 2004) closes with no error message. I'am using Repetier-Server Pro 0.94.3 on a Raspberry Pi 3.

Same behaviour on my Windows 10 PC with 2.3.0 beta1 and RepetierServerPro 0.94.3 on Raspberry Pi 4
PS closes without any message...

bubnikv, 2020-12-04:
@repetier Guys, could you please help out? Would you please help us debugging the Repetier issue? What are we doing wrong?

bubnikv, 2020-12-04:
@repetier
An excerpt for you: 

> PrusaSlicer crashes when clicking Update Printer.

This will hopefully be fixed in PrusaSlicer 2.3.0-beta2.

> No files are uploaded when after entering the host name and api code
> However when you insert "/printer/model/PRINTERNAME" in the field for the hostname, like
> https://repetierserver:portnum/printer/model/PRINTERNAME
> the gcode file is uploaded.

What is our poor slicer expected to do?

> There is another problem. It does not only upload the selected gcode under the name you enter in the dialog, but it also uploads a file which appeared to be a cachefile (eg .PrusaSlicer.upload.d46c-b4b0-f8c9-5bf4)

Now I have no idea.

repetier, 2020-12-04:
Ok, I have downloaded beta1 for mac so I could test directly with my development installation.
Now all my tries to upload a model worked. What I tested:
- Uploading to standard folder
- Uploading to named folder
- Names with non ascii chars
- Names with / - not good idea since it is path separator and everything before / was ignored, but apart from this successfull.
- Names with spaces worked.

All appeared in model folders as supposed to happen.

As a nice addition you could also allow uploading to queue or to queue and start print if none is running. Works the same just with tiny change in url:
- Add to queue:
url: /printer/job/<Your Printer Name>
extra parameter: autostart=false
- Add to queue and start print if none running: 
url: /printer/job/<Your Printer Name>
extra parameter: autostart=true

I would just add these 2 in the dropdown for groups and handle them as special case.

Note: In hostname I just entered 127.0.0.1:3344 not https (server normally is not reachable over https or if it has a private untrusted certificate so if you use correct uploading with certificate check it would fail). Only if server has public url you can get a valid certificate. A quick test with http://127.0.0.1:3344 also worked fine.

On the server I had user authentication activated and entered the global api key. That key is always available even without users and should always work.

Also note that the server it self has no https support. On the pi image the https solution uses nginx as proxy, therefore the portnumber can be omitted or it should be the regular https port :443 

Is your problem only with https since you mentioned:

> However when you insert "/printer/model/PRINTERNAME" in the field for the hostname, like
> https://repetierserver:portnum/printer/model/PRINTERNAME
> the gcode file is uploaded.

Or when exactly do you get problems. As said with beta 1 all seems to work in my tests so far. So seems you understood the api correctly.

repetier, 2020-12-04:
One more thing I noticed. In printer configuration you use in the dropdown the slug names of the printer. These are the internal names for referencing - so far good to use them. But externally the users always see the real names. When you use
/printer/info
to fetch the printer list it would be the "name" field that contains the visible printer name. Especially if it contains spaces or non ascii chars the two names differ.

bartwe, 2020-12-04:
Tested PrusaSlicer-2.3.0-beta1+win64-202011281033.zip
but it still crashes when pressing the 'Refresh Printers' button

wabbitguy, 2020-12-04:
I found that when connecting to a Raspberry PI4 I had to use a URL like 192.168.1.100 (example) but when connecting to a Repetier Server Pro running on a MacBookPro, I had to use a URL like 192.168.1.100:3344 or no connection could be made.  I did not preface either URL with http or https, strictly the IP.

In either case, the test worked perfect and populated the drop down "printer names" (slugs) as it should and I could upload directly to any of my four printers.

bubnikv, 2020-12-05:
> Tested PrusaSlicer-2.3.0-beta1+win64-202011281033.zip
> but it still crashes when pressing the 'Refresh Printers' button

That will be fixed in PrusaSlicer 2.3.0-beta2 (hopefully released today)

repetier, 2020-12-05:
@wabbitguy On the pi also nginx runs as proxy. Port 80 which is default port gets forwarded to 3344 same as 443 for https gets forwarded internally to 3344. On mac there is no nginx so you need to use the official 3344 port directly. On pi this would also have worked. nginx is just for the lazy and those needing https.

Dalton0190, 2020-12-05:
Hi, i can confirm that it works fine in beta2.
Thank you !

wabbitguy, 2020-12-05:
> @wabbitguy On the pi also nginx runs as proxy. Port 80 which is default port gets forwarded to 3344 same as 443 for https gets forwarded internally to 3344. On mac there is no nginx so you need to [...]

@repetier Excellent information to know. Thanks. Didn't find it in any of the documentation though so it might be a good idea to include someplace. Or how to use home-brew to install nginx...:-)

ropaha, 2020-12-05:
Works fine in beta2 for me too. Thx.

repetier, 2020-12-06:
@wabbitguy https://www.repetier-server.com/webserver-access-repetier-server/ also for linux it contains the required nginx config file to get it work on other os as well.

jmger, 2021-01-16:
> @docbobo Sorry for tagging you like this, but do you see what could the problem be upfront? I know that your PR was significantly altered before we merged it, but we tested and it seemed fine. Did [...]

> ### Version
> Version: 2.3.0-alpha3+win64
>
> ### Operating system type + version
> Windows 10 Professional 64bit
>
> ### 3D printer brand / version + firmware version (if known)
> MK3S MMU2S
>
> ### Behavior
> New [...]

I just found this thread on a google search. I have Repetier Server running on a Pi3. I was using Simplify3D to slice my projects. Uploading Gcode to Repetier server from my Mac Pro works great. I am trying out Slic3r, I get the same issue where the Gcode file would upload but not show up. After three tries I can finally get the Gcode file to show. Some times the gcode file shows up but never finishes to the point where I can see/press the print button. Finally Repetier Server just crashed with the boot loop "trying to connect to the server. Also I saw a post that this was a corrupt file. In trying to delete the bad gcode file, I see that there is no Repetier Server dir. I am not sure what happened. As I said repetier server was working great with the Simplify3D Gcode files.

repetier, 2021-01-16:
> I see that there is no Repetier Server dir
Which directory are you referring to? /var/lib/Repetier-Server ?  Server would recreate it if not there. But what you describe could be a corrupt sd card. Always use shutdown before unpowering pi or you get a defect sd card sooner or later where strange directory inconsistencies can happen. So check first if a new image would solve the problem since that will fix the directory structure hopefully. As far as I understand is the problem with wear leveling during shutdown. These errors are not seen by the ext 4 file system and not repaired on start up.

jmger, 2021-01-16:
Yes you are correct. The dir that I don't see is /var/lib/Repetier-server. The other info I found was to go to the dir /var/lib/Repetier-Server/printer/<printerslug name>/models. And check to see if there are any gcode files there that are corrupt and need to be deleted. But I don't see those directories. Yes, it could be a corrupt SD card. But I think it had something to do with the files that were produced by Slic3r. I never had this issue ever transfering files from Simplify3D. I guess I will have to reinstall Repetier Server and see if I get up and running again. But that does not account for Repetier Server not seeing my Slic3r gcode files. These 2 issues could be unrelated. But odd that this happened at the same time. I still think there is an issue with Slic3r slicing the gcode. One issue is that Slic3r is not adding an Extruder temp code. Or if it is, it is not telling the printer to start heating. I tried adding M109 S240. In in my start gcode right after the home command to see if that would work. I see the printer "Home" and the printer reading my BED Temp. (60c) and heats the bed. It never starts reading or heating the hotend. I put said hotend temp in, is it, "Print Settings" and I added M109 to my starting Gcode. But no go. At least it is pausing where it should be heating the hotend. Before it would just start printing without waiting for the hotend to reach temp.
I am new to Slic3r. I have been using Simplify3D for a few years. So any thoughts on Slic3r issues would be appreciated. P.S. The article said that the corrupt file problem was that a special character was being add or changing to when saving the file and that Repetier Server can't handle the filename. No way of knowing if that is true. One last thing, I click on Export Gcode in Slic3r and I assume it is exporting the correct gcode and that the gcode is being or already had been compiled. In my Version of Slic3r on the mac I do not have the option Creste Gcode now, or whatever it is cslled. I only have the option "Export Gcode". Next is to check my gcode file.

repetier, 2021-01-17:
/var/lib/Repetier-Server/printer//models is /var/lib/Repetier-Server/printer/<PrinterSlug>/models

When you have a printer configured the folder must exist. It gets created on startup. If not something is wrong with sd card.

I guess you use our doc for Simplify3D upload. PrusaSlicer uses exactly the same function for uploading only with the twist that you can select the target group. So as counter test check if Simplify3D uploading now also fails.

In PrusaSlicer printer definition your start gcode should contain somthing like this for temperature setting:

```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

mrcg, 2021-01-24:
I have tested again today with final PrusaSlicer 2.3.0.
The upload of gcode is still not working. Failure is still exactly the same as initially described.
Upload status shows "completed" but the file could not be found on Repetier Server.

When is it planned to make the feature working?

repetier, 2021-01-24:
Does the group have any special chars or so? I'm using it quite often now and had no problems so far. Using from mac prusaslicer to any server instance I have.

mrcg, 2021-01-24:
I somehow do not even understand the implementation to upload to special group.
I tried everything .... with/without backslash and frontslash. With/without beginning backslash and frontslash.
But without success.

Which string i need to add in the gcode upload window if i would like to upload a gcode (test.gcode) to Group (Group: Repair)?
It seems the usage is not self explaining and most probably i am doing something wrong. Thanks for your help.
Screenshots attached.
[image]
[image]

repetier, 2021-01-24:
Do not use any slashes at all. Just select the group. Only thing is I see between "Use forward ..." and Ok button a dropdown with the groups which you do not have. So question is if you have selected the repetier-server uploader or if that is the Octoprint version which has no groups to select from. That might also explain why it does not work.

See our tutorial made with windows:
https://www.repetier-server.com/knowledgebase/using-prusaslicer-with-repetier-server/

There you see the exact steps and there the window also looks different.

mrcg, 2021-01-24:
I have selected Repetier and entered the correct IP and API.
When pressing the Test button it says the connection is OK.
But i do not see any printer on the pulldown menu.
Any idea?
[image]

mrcg, 2021-01-24:
... and when i press the "Refresh Printers" button i get this error:

[image]

repetier, 2021-01-25:
Sounds like you did not use the api key. See our tutorial which uuid is the api key if you are unsure. But from the message the one you copied is wrong. Make sure to not copy spaces and make sue zero and O are correct. You can copy paste the api key.

mrcg, 2021-01-29:
I have followed your instructions multiple times. The api key is 100% correct (copy & paste without blank).
Test ... OK
Browse ... Button remains grey
Refresh Printer ... Button remains grey

I tried everything but it's impossible to connect.

mrcg, 2021-01-29:
After some kind of frustration i have re-installed the prusa slicer and now it's working!
Anyhow, thanks for your help and provided information.

closed 2021-01-29 by mrcg

Chazy12, 2021-02-11:
It's possible to start printing after the upload (as octoprint)?! I don't have the tick for reperire server..

repetier, 2021-02-11:
Currently not. I described on 20. December how it could be done without big changes but so far not included. Hope for next release they will add that.

mloidl, 2021-02-11:
Already requested in #5647. Ther is also a pending PR #5351.

Chazy12, 2021-02-11:
> Already requested in #5647. There is also a pending PR #5351.

That's sad. Mostly couse the repetier server is used for multiple printers..

Chazy