prusa3d/PrusaSlicer #5087: ctrl-I Import keyboard shortcut does not work

UltraSalem, 2020-11-05:
### Version
Version 2.3.0-alpha3+linux-x64-GTK3 Appimage

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Original prusa mk3s

### Behavior
Ctrl-I keyboard shortcut does not open an "import" file browser box to select an STL. Instead it spawns a whole extra instance of the PrusaSlicer GUI over the top of the existing one.
Ctrl-I should open an "Import" file browser box.

UltraSalem, 2020-11-05:
Seems like a new feature "open new Instance" in the Window menu has also been allocated ctrl-I. I have never needed to open a new instance through a keyboard shortcut. I use the Import shortcut at least once each time I use PrusaSlicer, but usually even more. Please give ctrl-I back to 'import'

rtyr, 2020-11-05:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4711. Closing.

closed 2020-11-05 by rtyr