prusa3d/PrusaSlicer #5088: Po otočení objektu chybí první vrstva

joka-cz, 2020-11-05:
PrusaSlicer
Version:   2.3.0-alpha3+win64
Build:     PrusaSlicer-2.3.0-alpha3+win64-202011041552

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,099MB
OpenGL installation
GL version:   4.3.0 - Build 10.18.15.4274
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 4600
GLSL version: 4.30 - Build 10.18.15.4274

### Behavior
Pokud otočím objekt o 45° podle osy X nebo Y, pak při slicování není vytvořena první vrstva, jakoby se objekt nedotýkal podložky.

#### Project File (.3MF) where problem occurs
přikládám 3MF a otisk obrazovky
[Error_first_layer.zip](https://github.com/prusa3d/PrusaSlicer/files/5492226/Error_first_layer.zip)
[image]

rtyr, 2020-11-05:
To je v pořádku. Při této orientaci je plocha modelu dole tak malá, že nejde standardní šířkou extruze vytisknout. Varovné hlášení na tuto situaci upozorňuje. Při použití parametru "Detect thin walls" se v tomto případě vytiskne i první vrstva.

joka-cz, 2020-11-05:
dobrý den,
máte pravdu, děkuji za vysvětlení.

closed 2020-11-05 by joka-cz