prusa3d/PrusaSlicer #5091: Crashes on Windows with 2 monitors

TheLazyHatGuy, 2020-11-05:
### Version
2.3.0-alpha3

### Operating system type + version
Windows 10 19041.610

### 3D printer brand / version + firmware version (if known)
N/a

### Behavior
* _Describe the problem_
PrusaSlicer crashes when using 2 monitors that have different resolutions.
* _Steps needed to reproduce the problem_
1. Launch the EXE on a non-primary monitor
2. Splash screen appears on main monitor and PrusaSlicer on the other monitor
3. Crash occurs
* _Expected Results_
PrusaSlicer runs
* _Actual Results_
PrusaSlicer crashes

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
N/a

TheLazyHatGuy, 2020-11-05:
I had a similar problem with 2.2.0 but the crash would only occur if I tried to slice on the other monitor

rtyr, 2020-11-05:
Looks like:
#4793
#2522
#2142
#3689

Nahimic driver issue?

TheLazyHatGuy, 2020-11-05:
I don't use the Nahimic Drive

bubnikv, 2020-12-07:
Do you still have the issue with PrusaSlicer 2.3.0-beta2?

TheLordStyle, 2020-12-07:
I experience this issue as well trying out 2.3.0 Beta 2.
Both of my monitors are the same resolution, when opening the app on the second (not primary monitor) the app hangs and crashes.

I had not launched the app at all, I had an explorer window open on the second window - launching the executable made the main Prusa screen show up on the second window but the splash screen show up on my main monitor.
The app hang and was non-responsive.
Killing the app and trying again - the issue still happened.

I quit the app a third time and moved the explorer window to my main monitor then launched the exe - the application then ran normally and I was greeted with the configuration wizard.

kocikdav, 2021-02-01:
Hello. I would like to ask users having troubles with Nahimic libraries to try this build. It should display warning dialog about NahimicOSD.dll linked.
https://www.dropbox.com/s/on0dwea75gfx4cc/PrusaSlicer-2.3.0%2B127-win64-gf939cac84-202102011345.zip?dl=0

We know that this is just a warning rather than a solution, but currently we have no way to stop loading this library as it comes linked statically when we start our process.

Also if you are aware of another libraries of same source please let us know their names. You can find it with ProcessExplorer tool.

TheLazyHatGuy, 2021-02-10:
Well, apparently SonicStudio3 uses the NahimicOSD.dll so I'll uninstall it and see if that fixes my problem.
[image]

TheLazyHatGuy, 2021-02-10:
Even though Sonic Studio 3 was disabled and never running it was apparently loading the libraries in the background whatever way a Windows App works because even though the files are still on my drive I no longer get the warning or the crash when using a second monitor.

Maybe this warning needs to be added to the main build if there is no way of getting around the library being loaded.

bubnikv, 2021-02-10:
@TheLazyHatGuy Thanks for testing. The Nahimic stuff may be bundled with various products I suppose.

> Maybe this warning needs to be added to the main build if there is no way of getting around the library being loaded.

This is exactly what we are planning to do.

TheLordStyle, 2021-02-12:
Thanks for the updates in this thread, I too had Sonic Studio 3 installed (bundled with my ASUS Motherboard but never used) removing the app as well as Sonic Radar 3 has fixed the issue for me, PrusaSlicer now runs on any monitor.

bubnikv, 2021-07-30:
Warning about the Nahimic DLL being injected into PrusaSlicer will be rolled out with PrusaSlicer 2.4.0-alpha1.
It is not a "fix" for the nahimic issue, but at least the user will be informed.

closed 2021-07-30 by bubnikv