prusa3d/PrusaSlicer #5094: [Feature Request] add gcode variable indicating which extruders are used

RossKor, 2020-11-06:
Provide a custom gcode variable indicating which extruders are used. Make this variable available to all custom gcode blocks.

I'm currently creating a custom profile for a printer with dual extruders and I'd like to create prime lines for the extruders which will be used in the print. Unfortunately, as far as I've been able to find, there is no way to tell which extruders are actually used.

*temperature[n] cannot be used as it always has a value because it is set by the selected filament and is not dependent upon the actual hotend temperature which will be set - I would find it acceptable to drive this value to zero for unused hotends*

Matts-Hub, 2020-11-07:
As per the [wiki documentation](https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)), you should be able to use the `current_extruder`, `previous_extruder` and `next_extruder` placeholders. Is this what you are wanting?

RossKor, 2020-11-07:
No. These are not indicators of which extruders will be used and next and previous extruder are not available to custom start gcode - [source](https://github.com/prusa3d/PrusaSlicer/blob/8491872fb78d6d2ae1d335ca0dcbfb3e3a06c063/src/libslic3r/GCode.cpp#L1377).

I've also noticed that if I only use extruder #2... both hotends get a temperature set - which is sort of problematic on its own.

RossKor, 2020-11-07:
I'm using Sailfish (MakerBot) gcode flavor by the way.

ardenpm, 2021-07-18:
This is definitely something I'd use on my Toolchanger. Right now I would need separate profiles with different start Gcode for every combination of potential used tools to avoid heating up the unused ones.

Dakkaron, 2024-05-27:
I have the same issue. My workaround was to create a filament called "Not used", which sets the temperatures to 0.

In the start gcode I select the highest value of all filaments for the bed temperature and only heat/prime hotends that have a temperature >0.

It would be really cool though if PrusaSlicer could do this without me having to manually select a fake filament since that's only one additional redundant step where I can mess up by selecting the wrong thing.

closed 2026-09-29 by prusaslicer-bot[bot]