prusa3d/PrusaSlicer #5095: Using modifier meshes to boolean difference areas of a model gives inconsistent results

Matts-Hub, 2020-11-06:
### Version
2.3.0 alpha3

### Operating system type + version
Windows 10 x64

### Behavior
I find that using modifier meshes as a way to subtract areas of a model that I don't want to be printed is extremely effective and useful, however it seems that it has some critical limitations.

When I use a modifier mesh to define an area where I want nothing to be printed, I set the modifier perimeters to zero, infill to zero, and top/bottom solid layers to zero. You can see that when this happens, the perimeters of the main body curve around the subtracted area to keep it sealed away - this is good. 
[image]
(cylinder shape is subtracted from the cube)

But this same logic is not applied to any horizontal faces that are left exposed by the subtraction, as you can see here:
[image]

and here:
[image]

I would expect these areas where you can see infill to be treated as top and bottom solid layers, and have solid infill there. I am aware that I am able to patch these gaps by making another modifier mesh to create 100% solid infill at the exposed areas, but the current implementation also has the consequence of not identifying areas where support material is needed *after* the subtraction, for example the image shown above, I could not get the slicer to put any support material there.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
You can see all the demo models screenshotted above in this project file
[demo (2).zip](https://github.com/prusa3d/PrusaSlicer/files/5498438/demo.2.zip)

Sembazuru, 2020-11-10:
This used to not be an issue (or I've forgotten my trick). I've used modifier meshes to leave voids for inserting objects or creating peg holes. After some version (I forget which) I stopped getting the horizontal layers above and below my modifier mesh voids.

bubnikv, 2020-12-07:
I think this is some special case.

bubnikv, 2021-06-03:
I think PrusaSlicer 2.4.0-alpha1 will handle your project somehow better as the handling of overlapping modifiers has been changed, so the attributes of overlapping modifiers are merged and the overlapping modifiers are applied in the order they are presented at the side panel (last wins).

[image]

Anyways, PrusaSlicer 2.4.0-alpha1 will implement real "negative volumes" aka "subtractors", which will be easier to use and they will work differently than the workaround that you were used to use: They will do a real 3D boolean before slicing.

closed 2021-06-03 by bubnikv