prusa3d/PrusaSlicer #5096: Why does the compilation fail when I follow the documentation How to build - Windows.md#Noob guide (step by step)? Is there another configuration?

sorrowfeng, 2020-11-06:


foreachthing, 2020-11-06:
Try this: https://github.com/prusa3d/PrusaSlicer/issues/5039#issuecomment-719985957

sorrowfeng, 2020-11-07:
Can you provide me with a compiled zip file, please?

sorrowfeng, 2020-11-08:
> Try this: [#5039 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/5039#issuecomment-719985957)

Why does slic3r\Release\libslic3r_gui.lib not compile, and the folder is empty?

foreachthing, 2020-11-08:
Please provide your build log, as far as it goes.

sorrowfeng, 2020-11-09:
Here's the build of the PrusaSlicer.

msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal

[log of prusaslicer](https://github.com/sorrowfeng/share/blob/main/log_of_prusaslicer.txt)

Here's the build of the deps.

msbuild /m ALL_BUILD.vcxproj

[log of deps](https://github.com/sorrowfeng/share/blob/main/log_of_deps.txt)

sorrowfeng, 2020-11-09:
> Please provide your build log, as far as it goes.

Here's the build of the PrusaSlicer.

msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal

[log of prusaslicer](https://github.com/sorrowfeng/share/blob/main/log_of_prusaslicer.txt)

Here's the build of the deps.

msbuild /m ALL_BUILD.vcxproj

[log of deps](https://github.com/sorrowfeng/share/blob/main/log_of_deps.txt)

foreachthing, 2020-11-09:
Ok .. so, if I compare your log (left) to mine (right), I can see one clear difference:
[image]

`"C:\dev\PrusaSlicer\PrusaSlicer\build\PrusaSlicer.sln"` -> you
`"C:\dev\PrusaSlicer\build\PrusaSlicer.sln"` -> me

You're building in one directory too deep. My guess is, that my script then fails because it assumes `C:\dev\PrusaSlicer\build\`. Try to fix that and post your results again.

sorrowfeng, 2020-11-09:
> Ok .. so, if I compare your log (left) to mine (right), I can see one clear difference:
> [image]
>
> `"C:\dev\PrusaSlicer\PrusaSlicer\build\PrusaSlicer.sln"` -> you [...]

I recompiled it, and in C:\dev\PrusaSlicer\, here is the log output for the PrusaSlicer compilation.

[log output](https://github.com/sorrowfeng/share/blob/main/log.txt)

Can you provide me with a configuration file for visual studio installer? I'll see if there is a problem with my installation, thanks.

sorrowfeng, 2020-11-10:
> Ok .. so, if I compare your log (left) to mine (right), I can see one clear difference:
> [image]
>
> `"C:\dev\PrusaSlicer\PrusaSlicer\build\PrusaSlicer.sln"` -> you [...]

Can you use google drive to provide me with a compiled archive, bro?

foreachthing, 2020-11-10:
> Can you use google drive to provide me with a compiled archive, bro?

Latest build: [https://file.io/FZL1OyQI0sKz](https://file.io/FZL1OyQI0sKz)
This is a ONE-TIME download! Once you've downloaded it, please let me know, so I can remove this link.

sorrowfeng, 2020-11-10:
> > Can you use google drive to provide me with a compiled archive, bro?
> 
> Commit #70b19d2
> https://gofile.io/d/eDPV79
> File expires on: Mon Nov 16 2020 00:00:00 GMT+0100

I've finished the file, but it's not what I wanted.
I want that big file that's 10+ GB after compilation.

Can you give me your log output file, I'd like to compare it myself.
Here is the full log output file after my bat file is executed.
[log output](https://github.com/sorrowfeng/share/blob/main/all_build_bat_log.txt)

foreachthing, 2020-11-10:
Here's my build log. I hope this helps!
[PS-Build-Log.log](https://github.com/prusa3d/PrusaSlicer/files/5517742/PS-Build-Log.log)

lukasmatena, 2020-11-10:
Guys, I will close this. As far as I can tell, this is not an issue with PrusaSlicer and probably not even with its cmake build scripts.
You can of course discuss all you want, but the issue should be closed, so it does not clutter the list of open ones. I hope that's fine.

closed 2020-11-10 by lukasmatena

sorrowfeng, 2020-11-11:
> Here's my build log. I hope this helps!
> [PS-Build-Log.log](https://github.com/prusa3d/PrusaSlicer/files/5517742/PS-Build-Log.log)

Do you have a log of the compiled deps?

foreachthing, 2020-11-11:
No, not anymore. 
I would have to recompile the while pile from scratch... when I get to do it, I'll let you know.

sorrowfeng, 2020-11-11:
> No, not anymore.
> I would have to recompile the while pile from scratch... when I get to do it, I'll let you know.

Thank you so much.

foreachthing, 2020-11-11:
I'm rebuilding right now... meanwhile: did you run that batch in cmd or "Developer PowerShell"?

foreachthing, 2020-11-11:
Here's the full log! It's about 15 MB in size ... 😮 
[fulllog.zip](https://github.com/prusa3d/PrusaSlicer/files/5527131/fulllog.zip)

sorrowfeng, 2020-11-12:
> I'm rebuilding right now... meanwhile: did you run that batch in cmd or "Developer PowerShell"?

I used  Developer PowerShell.

sorrowfeng, 2020-11-13:
> Here's the full log! It's about 15 MB in size ... 😮
> [fulllog.zip](https://github.com/prusa3d/PrusaSlicer/files/5527131/fulllog.zip)

I've reinstalled the system and compiled it in a clean environment, but it still fails in the project libslic3r_gui, can you give me some advice?

[image]

[image]

foreachthing, 2020-11-13:
Well, to honest, I have no clue why this would not compile!
Then again, I don't use VS directly. I compile from command line (should not matter, I guess).

Have you tried a clean new computer? I use virtual computers (VirtualBox) for all my dev work, that way I could reset a machine without loosing all my data (which is in a directory mapped to the virtual computer).
There is something odd on your computer...

sorrowfeng, 2020-11-13:
> 好吧，说实话，我不知道为什么不编译！
> 再说一次，我不直接使用VS。我从命令行编译（我想应该没关系）。
> 
> 您尝试过一台干净的新计算机吗？我在所有开发工作中都使用虚拟计算机（VirtualBox），这样我就可以重置计算机而不会丢失所有数据（位于映射到虚拟计算机的目录中）。
> 您的计算机上有些奇怪的东西...

I reinstalled my laptop and tried installing only VS2019 and git, but still get the same error.

I also use command line compilation, and use VS just to help me locate the error location.

Did I miss some components when I installed VS, can you share with me the components you installed?

foreachthing, 2020-11-13:
I have the Pro version ... I don't know how much help this would be. And, I don't thing you need everything I have.

[image]

Have you tried to set your OS language to English (if not already)?

sorrowfeng, 2020-11-14:
I reinstalled an English version of windows and it solved the problem, is it the coding?
Thank you very much for your help.

foreachthing, 2020-11-14:
>.. it solved the problem, is it the coding?

Must be it!! Glad you got it working! 
:-)