prusa3d/PrusaSlicer #5099: Monotonic top infill produces mixed results

snowdogg00, 2020-11-06:
### Version
2.3 Alpha1

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10_64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3 with bondtech geared extruder, fw 3.9.0

### Behavior
Top layer from one object comes out clean, second object during same print produces a seam-like structure

[image]
[image]

rtyr, 2020-11-06:
Can you please provide your 3MF project file?

snowdogg00, 2020-11-06:
[glyph_6.zip](https://github.com/prusa3d/PrusaSlicer/files/5503852/glyph_6.zip)

Matts-Hub, 2020-11-07:
Looking at your project file, it appears as though these lines are due to the layer below the top monotonic infill layer. It only appears on one of the objects because that layer on that object is broken up by holes in it, so it completes the infill in sections. Where these sections meet, corresponds with the lines you see on your top layer. See here, layer 29:
[image]

Perhaps some under or over extrusion in those areas is responsible, but it's definitely not the fault of the monotonic infill. I suspect if you reprint this object with regular rectilinear infill you will see the same result.

Sembazuru, 2020-11-10:
I've noticed this before on other models. Maybe if the internal top layers were also printed monotonic it would help. With this example model, the one on the left was printed with rectilinear and the one on the right printed with monotonic (g-code generated in alpha2). Both show a ridge in the same place from the second to top layer, but the surface discontinuities and ridge in the top surface were corrected by the monotonic top layer fill.
[image]

For reference, here is the (zipped) .3mf file that I used to generate these two. (I was using it to experiment with ironing on rectilinear and monotonic fills. Source files also included in the zip file.)
[FatesEndPegFilingFixtures.zip](https://github.com/prusa3d/PrusaSlicer/files/5515807/FatesEndPegFilingFixtures.zip)

rtyr, 2020-11-12:
Related to https://github.com/prusa3d/PrusaSlicer/issues/4425.

bubnikv, 2020-12-07:
Closing this one, keeping https://github.com/prusa3d/PrusaSlicer/issues/4425 open

closed 2020-12-07 by bubnikv