prusa3d/PrusaSlicer #5101: Add ability to have different density of support material for first layer

VorpalBlade, 2020-11-07:
### Version
2.3.0-alpha3+linux-x64 (GTK 2 version)

### Operating system type + version
Ubuntu 18.04 x86-64

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3s with powder coated PEI sheet. Firmware version 3.9.1.

### Behavior
During certain prints with PETG (non-prusament brand: Prima3D EasyPrint PETG, white) for certain parts with unfortunate geometry, I experienced bed adhesion problems with the supports. I added a brim and this kept the part from lifting totally. 

However, as can be seen in the photo, the supports still lift slightly from the bed. I have tried everything except glue stick (which should not be used for powder coated PEI as I understand it). As can be seen the brim is well affixed.

[image]

As the parts are replacements for broken injection moulded pieces, changing geometry is not possible (as they have to fit in a tight space in a car). It is also difficult to change printing orientation as the parts will be mechanically loaded and I want to have the strongest direction horizontally as printed.

Thus my idea is that it should be possible to have less space between the support "lines" on the first layer, basically making the brim extend under the supports, all the way to the part. As I want the support to be easy to remove and not waste plastic I don't want to increase the over-all density of the supports. As far as I can tell this is not currently possible.

_Is this a new feature request?_
Yes.

#### Project File (.3MF) where problem occurs
[xc70_cargo_cover_handles.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5504620/xc70_cargo_cover_handles.3mf.gz)

tbrummel, 2020-12-05:
Your idea of printing the first support layer with its own spacing is interesting - a great idea I think.  I was about to file a bug report for the current behavior -- the bug being that the first layer of support does not respect the "Support pattern spacing" parameter.  Why not?  Is this not a bug?

Still, I like your idea better than just fixing this -- I'm attaching an example with the support spacing set to 0 (solid) to make it easy to see.  Notice how the first layer of support material being printed is useless?  It's not touching anything and does not help keep the supports on the bed.  I've tried everything I can think of to print that first layer as solid (or to get it to follow my support spacing), but it doesn't seem possible.  Your idea would allow making the first layer solid (great bed adhesion for a small cost in material) and other layers the desired pattern density.

[image]

bubnikv, 2021-02-24:
Duplicate of #756, #2101, #1165, #1445 etc.
Implemented with fcb714cd241f0847522b9aac57da6b62acdd0152

closed 2021-02-24 by bubnikv