prusa3d/PrusaSlicer #5103: Anycubic Printers Rename

Flo082002, 2020-11-07:
Hi,
the Anycubic printers have a wrong name.
[image]

rtyr, 2020-11-07:
The names are correct, but the wizard is displayed incorrectly on your system for some reason. Can you please share some details about your system?

[image]

rtyr, 2020-11-07:
Possibly related to https://github.com/prusa3d/PrusaSlicer/issues/4861.

Flo082002, 2020-11-07:
Ubuntu 20.04
PrusaSlicer-2.3.0-alpha3+linux-x64-GTK3-202011041604

rtyr, 2020-11-08:
I was able to reproduce it on Ubuntu 20.04, but only at first aplication start (PrusaSlicer-alpha data folder didn't exist). It is OK since second PrusaSlicer start.

Flo082002, 2020-11-08:
Same for me

pmjdebruijn, 2020-11-10:
> I was able to reproduce it on Ubuntu 20.04, but only at first aplication start (PrusaSlicer-alpha data folder didn't exist). It is OK since second PrusaSlicer start.

Keep in mind that 90% of the Wizard's usecase is first application start.

This is indeed a duplicate of #4861

rtyr, 2020-11-10:
> Keep in mind that 90% of the Wizard's usecase is first application start.
> 
Sure, I just reported my results, I don't think it is OK like that.

pmjdebruijn, 2020-12-02:
This issue still persist in current master.

YuSanka, 2020-12-17:
Thanks for your report. This bug was fixed and will be a part of upcoming release.

closed 2020-12-17 by YuSanka