prusa3d/PrusaSlicer #5105: No support with 0.2mm for custom supports by painting

Shama71, 2020-11-07:
### Version
2.3.0-alpha3+win64

### Operating system type + version
Windows 10

### Behavior
Custom supports by painting no support with 0.2mm.
For other layer height support si correct.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
Is correct
[image]

No support
[image]

[Bug - Support.zip](https://github.com/prusa3d/PrusaSlicer/files/5505577/Bug.-.Support.zip)

lukasmatena, 2020-11-08:
This also appears when using good-old custom box enforcers, so it is not a bug in the new painter. Even a small change from 0.2 mm fixes it.

bubnikv, 2020-12-11:
Maybe you have "dont't support bridges' enabled?

Shama71, 2020-12-11:
> Maybe you have "dont't support bridges' enabled?

Yes it is included, but sometimes support was needed for large bridges, but not needed for normal ones.

bubnikv, 2020-12-11:
Currently the support enforcers (painted or boxes) are canceled by the
"don't support bridges" feature. We may change it in the future, but for
now this is what we have. We will not be able to change the behavior in
PrusaSlicer 2.3.0 release.

minosg, 2021-01-18:
@bubnikv 

Out of curiosity, what would be the design requirements for that logic? 

I understand the logic behind the automatic support planner not generating support for every bridge feature. But, if a user wants to print a cup type object, and using "Support enforcers only", explicitly define his own supports, why would this option stop him from doing so?

bubnikv, 2021-01-19:
> Out of curiosity, what would be the design requirements for that logic?

It was not by design, it was by implementation. The implementation of support enforcers was significantly simpler that way. We may change the behavior in the future.

minosg, 2021-01-19:
@bubnikv thanks for the prompt response. That makes more sense. Since it is not hard to workaround by disabling that checkbox, it shouldn't be a high priority task.

There may be some value in disabling that box in the default profiles though, or adding some context in the tool tip, to avoid duplicate ticket/bug reports from new users.

bubnikv, 2021-03-18:
The support enforcers newly take precedence over "don't support bridges" and "on build plate only".

[image]

Shama71, 2021-03-18:
Actually, in the first message, I pointed out that when the height of the layer changes, the behavior of the support placement changes.

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv