prusa3d/PrusaSlicer #5107: Separate Tool Print Settings

Design-Create-Innovate, 2020-11-08:
This is a feature I've been hoping they would implement for a while.
Obviously I know this software is for prusa printers but seeing as many (like myself who switched from simplify3D) use PrusaSlicer for a 3rd party printer it would be nice to have more control for dual extrusion that uses two separate extruders. The main problem I have is that I have a 0.8mm nozzle on one extruder and the other with a hardened 0.4mm. I can use them individually but if I want to print a dual material print it wouldn't pull the setting I have tuned for one of the extruders. So in short, I would just like to see that added in a fashion similar to the option given for multiple filaments on the sidebar.

rtyr, 2020-11-08:
It seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/393.

Design-Create-Innovate, 2020-11-09:
oh i suppose it is kind of a duplicate of #393, but is that being addressed then?

bubnikv, 2020-12-07:
Closing this one, keeping https://github.com/prusa3d/PrusaSlicer/issues/393 open.

closed 2020-12-07 by bubnikv