prusa3d/PrusaSlicer #5108: Wrong button/textfield placement

Flo082002, 2020-11-08:
### Version
PrusaSlicer-2.3.0-alpha3+linux-x64-GTK3-202011041604
Ubuntu 20.04
Language: German

Textfields are on top of every  + und - button

Some examples:
[image]
[image]
[image]
[image]
[image]

fe60, 2020-11-10:
Similar to #5053

pmjdebruijn, 2020-11-10:
This is a duplicate of #5062

pmjdebruijn, 2020-11-11:
Probaby solved by a2b915af0fafd7a52758dcef6f45647953ff3bd5

YuSanka, 2020-11-12:
Close as duplicate of #5062 and #5081

closed 2020-11-12 by YuSanka