prusa3d/PrusaSlicer #5111: Support artifacts due to (maybe) supports going through wall?

CobaltEcho, 2020-11-08:
### Version
2.3.0-alpha3+win64
(Also on 2.2+, which is the first I used.  Tried 2.3.0-alpha3 to see if it made a difference (it did not))

### Operating system type + version
Win7

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Weird artifacts with supports.  Painting blockers does not fix.  You can see how there are supports all over the back to the eagle.  Not limited to this file.  In this example, the support angle the is the same in Cura and PrusaSlicer.

It almost seems like the supports are going trough the object and coming out the other side.  Shouldn't the supports stop at the object wall?

PrusaSlicer Pic:
[image]

Cura Pic (teal color are supports):
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
This is a different 3mf file from the pics above (I didn't save 3mf), but this has the same thing going on, with some pics below:
3MF: [Middle Finger Parts.zip](https://github.com/prusa3d/PrusaSlicer/files/5507242/Middle.Finger.Parts.zip)

Pics from 3MF:
[image]
[image]

bubnikv, 2020-12-08:
The artifacts on the middle finger are columns trying to support what happens inside the hole above.

I would be interested in the bird model though, the supports are extreme there. Would you please attach a 3MF of it?

CobaltEcho, 2020-12-10:
@bubnikv , thanks for exsplaining about the holes.  For some reason this 3MF file is huge (and it takes a long time to slice).  I imagine its all the feathers on the wings.  I had to upload it to google drive since it's over 10MB.

https://drive.google.com/file/d/1hFACR6usYXDbmoW49rvhpVOKj3JxQRt1/view?usp=sharing

bubnikv, 2020-12-11:
Thanks for the model.

Neither Cura handles it nicely. It is true that Cura creates less volume of supports, but the question is which one will support better and leave less scars.
[image]

The problem here is that your model requires a lot of supports when looking at the details at close up. Neither PrusaSlicer nor Cura evaluate the need for support globally, they both use a local algorithm. PrusaSlicer filters some of the small overhangs and I suppose Cura does that as well, and maybe Cura does it a little better? Still neither filters out those islands, that would better not be supported as the supports would be difficult to remove and drooping extrusions would be all right.

bubnikv, 2020-12-11:
The main difference between PrusaSlicer and Cura in the case of the Eagle is that PrusaSlicer extends the support interfaces to a regular grid to make them more stable, while Cura just smoothens the boundaries a bit.

bubnikv, 2021-10-05:
The situation is somehow better with the snug supports of PrusaSlicer 2.4.0-alpha3.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]