prusa3d/PrusaSlicer #5112: [2.3.0-alpha3] Seam painting crashes on slicing

yincrash, 2020-11-08:
### Version
_2.3.0-alpha3_

### Operating system type + version
_macOS 10.15.7

### 3D printer brand / version + firmware version (if known)
_Ender 3 Pro with Duet and modified with a 2nd extuder (single hotend)_

### Behavior
* App crashes when attempting to slice with seam painted
* Stack trace for the thread that crashes - 
```Thread 2 Crashed:: slic3r_BgSlcPcs
0   com.prusa3d.slic3r/           	0x000000010e7dabb5 Slic3r::Polygon::contains(Slic3r::Point const&) const + 5
1   com.prusa3d.slic3r/           	0x000000010e5dbb46 Slic3r::ExPolygon::contains(Slic3r::Point const&) const + 22
2   com.prusa3d.slic3r/           	0x000000010e677ef8 Slic3r::SeamPlacer::get_enforcers_and_blockers(unsigned long, Slic3r::Polygon const&, std::__1::vector<unsigned long, std::__1::allocator<unsigned long> >&, std::__1::vector<unsigned long, std::__1::allocator<unsigned long> >&) const + 168
3   com.prusa3d.slic3r/           	0x000000010e677050 Slic3r::SeamPlacer::apply_custom_seam(Slic3r::Polygon const&, std::__1::vector<float, std::__1::allocator<float> >&, std::__1::vector<float, std::__1::allocator<float> > const&, int, Slic3r::SeamPosition) const + 144
4   com.prusa3d.slic3r/           	0x000000010e676a84 Slic3r::SeamPlacer::get_seam(unsigned long, Slic3r::SeamPosition, Slic3r::ExtrusionLoop const&, Slic3r::Point, double, Slic3r::PrintObject const*, bool, Slic3r::EdgeGrid::Grid const*) + 3236
5   com.prusa3d.slic3r/           	0x000000010e6b97a9 Slic3r::GCode::extrude_loop(Slic3r::ExtrusionLoop, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, double, std::__1::unique_ptr<Slic3r::EdgeGrid::Grid, std::__1::default_delete<Slic3r::EdgeGrid::Grid> >*) + 377
6   com.prusa3d.slic3r/           	0x000000010e6ba0da Slic3r::GCode::extrude_entity(Slic3r::ExtrusionEntity const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, double, std::__1::unique_ptr<Slic3r::EdgeGrid::Grid, std::__1::default_delete<Slic3r::EdgeGrid::Grid> >*) + 554
7   com.prusa3d.slic3r/           	0x000000010e6bb857 Slic3r::GCode::extrude_perimeters(Slic3r::Print const&, std::__1::vector<Slic3r::GCode::ObjectByExtruder::Island::Region, std::__1::allocator<Slic3r::GCode::ObjectByExtruder::Island::Region> > const&, std::__1::unique_ptr<Slic3r::EdgeGrid::Grid, std::__1::default_delete<Slic3r::EdgeGrid::Grid> >&) + 311
8   com.prusa3d.slic3r/           	0x000000010e6b6678 Slic3r::GCode::process_layer(__sFILE*, Slic3r::Print const&, std::__1::vector<Slic3r::GCode::LayerToPrint, std::__1::allocator<Slic3r::GCode::LayerToPrint> > const&, Slic3r::LayerTools const&, std::__1::vector<Slic3r::PrintInstance const*, std::__1::allocator<Slic3r::PrintInstance const*> > const*, unsigned long) + 15496
9   com.prusa3d.slic3r/           	0x000000010e6ad57e Slic3r::GCode::_do_export(Slic3r::Print&, __sFILE*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 23134
10  com.prusa3d.slic3r/           	0x000000010e6a6105 Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 1045
11  com.prusa3d.slic3r/           	0x000000010e860ae3 Slic3r::Print::export_gcode(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 755
12  com.prusa3d.slic3r/           	0x000000010eb6747a Slic3r::BackgroundSlicingProcess::process_fff() + 426
13  com.prusa3d.slic3r/           	0x000000010eb69b46 Slic3r::BackgroundSlicingProcess::thread_proc() + 262
14  com.prusa3d.slic3r/           	0x000000010eb6b3e2 boost::detail::thread_data<Slic3r::BackgroundSlicingProcess::start()::$_1>::run() + 18
15  com.prusa3d.slic3r/           	0x000000010f04c498 boost::(anonymous namespace)::thread_proxy(void*) + 120
16  libsystem_pthread.dylib       	0x00007fff71851109 _pthread_start + 148
17  libsystem_pthread.dylib       	0x00007fff7184cb8b thread_start + 15
```

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Gopro_mount_Canopy.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5507393/Gopro_mount_Canopy.3mf.zip)

yincrash, 2020-11-09:
Might be a dupe of https://github.com/prusa3d/PrusaSlicer/issues/5014

lukasmatena, 2020-11-27:
Indeed a duplicate of #5014, the crash was caused by raft that was unaccounted for. Fixed with f485f66 for PrusaSlicer 2.3.0-beta. Closing.

closed 2020-11-27 by lukasmatena