prusa3d/PrusaSlicer #5113: Ironing - disable retract after ironing

Wireheadbe, 2020-11-09:
### Version
Version 2.3.0-alpha3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 - Microswiss DD - Klipper

### Behavior
After ironing - the default retraction might be to much. It's causing underextrusion in the subsequent moves. Suggest adding an option "no retract after ironing" or disabling retraction.

#### Project File (.3MF) where problem occurs
[point_8_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5509319/point_8_cube.zip)

See below image - left sliced with Ironing all top surfaces - right, no ironing. You can clearly see the underextrusion after the retraction happens after ironing.
[image]

bubnikv, 2020-12-07:
I think your issue is the same as 
https://github.com/prusa3d/PrusaSlicer/issues/5000#issuecomment-740045845

the problem is caused by large differences in nozzle pressure.

michael-kerbel, 2020-12-13:
I am experiencing the same issue with ironing. Honestly I don't understand the explanation of nozzle pressure or what that could mean regarding a fix or a workaround. Right now, the feature is only usable for the top-most surfaces, otherwise the prints are basically broken on that layer.

michael-kerbel, 2020-12-15:
Disabling retraction did not solve the issue for me. But since it was a temporary jam in the extruder, I was able to fix it by increasing the print fan speed as a workaround. That seems to prevent the extruder getting too hot due to the low flow of filament in such a long period of time.
This is a before/after picture:
[image]

bubnikv, 2020-12-16:
There are two issues:
1) Heat creep is quite fatal, it makes the filament soft in the heat break and it may lead to print fail or missing layers before the plug gets pushed through. Some filaments may be more sensitive to heat creep, for example those filled with mineral fillers (talc, kaolin clay, titanium dioxide)
2) Pressure difference. This may get compensated by linear advance algorithm in the firmware, but in general FDM printers love to print with a constant volumetric speed.

michael-kerbel, 2020-12-16:
Thanks! „Heat creep“ is a very fitting term. I definitely had that.

What is not visible in my picture is, that I get a „fat“ perimeter on the layer after the ironing. But my g-code already includes the linear advance command. I wonder if printing the infill first after the ironing layer would be helpful to mitigate that.

michael-kerbel, 2020-12-17:
I hacked the gcode to print the infill first, but that was simply not massive enough to make a difference. So I added an additional object to the print, with a 100% infill layer right after the ironed layer. That way, the fat inconsistent extrusion happens in the trash object.
[image]

That worked very well. Now I get ironing with perfect layers. To the left you see the fat layer, to the right the result of my workaround:
[image]

Unfortunately I have no idea how to add that workaround to PrusaSlicer. I hope someone comes up with a better idea how to fix this.

TheRealMozes, 2020-12-26:
I have the same issue of getting a fat layer after ironing. I've tried different materials and filaments and played with the ironing settings nothing seems to help. Odly enough though, i've printed 12 of those at once all with ironing and here some of them didn't have the fat layer. 

[image]

michael-kerbel, 2020-12-26:
Printing multiple objects at once is basically my workaround, with using the real objects as "trash object".

TheRealMozes, 2020-12-26:
My objects are a bit of a hard one as it has three not connected surfaces on the same layer from the same object that need to be ironed. It does the perimeters, infill, irons it and then goes to the next. 

[image]

This way my part always has two injections of a fat layer. Three even if the previouse move on the prev part was an iron. So it isn't a solution for me. It prob can only be really fixed by printing everything on a layer, then ironing it all with somekind of waste object after each iron move alla MMU2. I also want to print a bunch of these objects at once, not just one and a waste object at a time. Maybe there is a way to get around it in a different way idk. 

I really hope this gets looked after as ironing can really make your prints look more like they are injection molded. My girlfriend commented that my parts looked like they were from a shop :).

michael-kerbel, 2020-12-28:
I see, this is tough indeed. As a workaround for the workaround you could manually edit the gcode and move the ironing of the two surfaces to the same block, where the ironing for the last surface happens. I did that in my tests, it is pretty easy and you can check whether it will do what you want to by loading the gcode into the gcode viewer of Slicer. Then the waste object would work again as a workaround.
But I really hope someone has a better idea, since this is obviously not viable as a real solution.

closed 2026-09-29 by prusaslicer-bot[bot]