prusa3d/PrusaSlicer #5115: Wipe into infill functionality erratic

Pun-e, 2020-11-09:
### Version
2.2.0 Final
2.3.0-A3

### Operating system type + version
W10

### Behavior
The "wipe to infill" option seems very unpredictable in its functionality. On some layers it works as expected (note the roughly 20% of the purge tower that's simply empty space)
[image]

But then a few layers later:
[image]

I thought that maybe this bug had been addressed in the latest beta, but it's there.

Using [this great design](https://www.thingiverse.com/thing:2485063)

lukasmatena, 2020-11-09:
Please post a 3MF and give us the exact layer number where you think is a problem. Thanks.

Pun-e, 2020-11-10:
In this file, **only** the cell walls have wipe to infill (WTI) turned on.

For layers 123 - 165 the main body has WTI disabled because there is another non-connected object being printed with the same extruder.

For layers 203 -229  the main body has WTI disabled because there is a connected object being printed with the same extruder

The second case is potentially understandable but still undesirable, and the first case cannot be anything but a bug

[Wipe to infill issues.zip](https://github.com/prusa3d/PrusaSlicer/files/5515253/Wipe.to.infill.issues.zip)

lukasmatena, 2020-11-10:
@Pun-e Thanks. I agree that this is a bug. I tried to slice the same 3MF in PrusaSlicer 2.1.1 and the results were not the same. in 2.1.1, all layers above 21.8 mm (number 109) up to 51.6 mm (258) use the big part for infill-wiping. Starting with 51.8 mm (259) it happens every other layer (this is intentional, infill must be printed after perimeters, so when you have just two extruders...). You can try it out and tell us if that would make you happy.

Version 2.2.0 is already broken. Serious refactoring of the wipe into infill feature took place shortly before 2.2.0 release, and it broke it in several places. We hoped that all were fixed before 2.2.0. This may be one that we overlooked.

vintagepc, 2020-11-10:
Note that the WTI feature does not split infill to wipe, so if you only have one large infill volume, it appears to still only wipe a single colour into that area, even if the available volume is sufficient for additional purges.

Pun-e, 2020-11-11:
@vintagepc Ah, now that explains a lot.

@lukasmatena So you want me to install 2.1.1 and see if it's all behaving as I'd expect / like?

I gotta say vintagepc's point is one I would expect a consumer-ready MMU slicing to be able to handle. If I have 500mm^3 of infill, it's a bit silly that the slicer can't figure out how to do multiple purges within it even if the changeover itself must be done over the purge tower.

I think using "no sparse layers" with combined infill every second layer could offer a big improvement in material efficiency and a slight increase in print speed, but right now that combination isn't working as expected.

I'm sorry if this is getting off base with my original point, would you rather I made a new ticket for each of my small issues with Prusa slicer and the MMU or expand on them here?

vintagepc, 2020-11-11:
I think it is due to the way the wiping is calculated - first all of the toolpaths are generated, and then, if WTI is checked, it will check the continuous paths it is able to use for wipe on the next layer and assigns them one by one. 

Based on my limited poking in the PS code, I don't think it's technically impossible, but I can also understand why it is not a trivial task to "split" a toolpath for this so far down the generation process. It's a bit of a catch 22 since at the time you are generating the paths, you likely are not ready to deal with tool assignments yet, but by the time you have that info ready, it's much harder to change the paths. 

I wonder if it could be worked around/kludged by using the "object clipping" behaviour to add internal dummy "parts" in large infill areas to break up the paths and purging. 

At least in my case I tend to have something standing by to use as a wipe object rather than rely purely on WTI alone. Modular storage drawer type parts are excellent wipe objects, since you generally don't care too much how they look and generally one always has a use for more storage drawers for parts and pieces. :smile:

lukasmatena, 2020-11-11:
@Pun-e 
> So you want me to install 2.1.1 and see if it's all behaving as I'd expect / like?

I'm not saying you must. But if you want to, yes.

> would you rather I made a new ticket for each of my small issues with Prusa slicer and the MMU or expand on them here?

If you have something else, definitely open a separate issue (after searching if it wasn't discussed before).

> it's a bit silly that the slicer can't figure out how to do multiple purges within it

It is. It is also silly that the G-Code export is not parallelized, that you can't paint color on the objects, that there is no serial controller, adequate GTK3 support, better support generator, tree supports etc. It is about priorities. Something like the 80/20 rule. Improving WTI further is possible, but it would cost a lot of time and bring additional benefit in limited number of cases (those with very large islands). Maybe one day. Until then you can use a workaround described in https://github.com/prusa3d/PrusaSlicer/issues/2822#issuecomment-524874646. Now I'm more worried about the bug that you described, though.

@vintagepc 
> at the time you are generating the paths, you likely are not ready to deal with tool assignments yet

That's the problem. Not that it is not technically possible, but it is not simple to do. The software evolves continuously and some changes are then harder than it seems from the outside, because nobody expected you'd need to do such things. Some things are already hacked around, because you can't change the whole architecture with every new idea.

closed 2026-09-29 by prusaslicer-bot[bot]