prusa3d/PrusaSlicer #5116: Ironing Brush Selector

luckyguido, 2020-11-09:
### Version
PrusaSlicer 2.3.0-alpha3

### Operating system type + version
Windows 10 (build 19042), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3 3.9.1

### Behavior
Since brush tool and ironing features have been developed and work so nice, it will be great to use them togheter using brush to select wich surfaces need to be ironed.
I can't figure more options on ironing drop down but in many cases they may apply ironing too much or not enough.
Ironing feature is quite slow during printing process and optimizing it by reducing ironed surface with brush selection will make print still look better and save a lot of time!

_Is this a new feature request?_
YES!

fe60, 2020-11-18:
I like the idea of having an option to define the surfaces, that shall be smoothed. But I don't like the brush tool.
Instead I would like to see a more precise option to control the ironing, like:
- "Ironing Blockers/Enforcers" similar to the "Support Blockers/Enforcers", or
- a list with layers having horizonatal surfaces and an option to select the ones that shall be ironed.

bubnikv, 2020-12-07:
> "Ironing Blockers/Enforcers" similar to the "Support Blockers/Enforcers", or
> a list with layers having horizonatal surfaces and an option to select the ones that shall be ironed.

Both work already, just apply the infill modifiers.

bonjipoo, 2020-12-23:
+1 As ironing is very time consuming it would be good to be able to selectively apply it to a model.

bubnikv, 2021-01-05:
Ironing could be enabled / disabled in a layer range modifier or in a modifier mesh.

closed 2026-09-29 by prusaslicer-bot[bot]