prusa3d/PrusaSlicer #5119: Crash on Repetier Server "Refresh Printers" Button

whackl, 2020-11-10:
### Version
PrusaSlicer 2.3.0-alpha3+linux-x64
### Build 
PrusaSlicer-2.3.0-alpha3+linux-x64-202011041552

Same Problem on .Appimage and GTK3 .AppImage and tar.bz 

### Operating system type + version
Ubuntu 20.04 LTS 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s 3.9.1

### Behavior
* Slicer Crashes after Clicking Refresh Printers Button

* Select Host Type Repetier
* Fill in IP of Repetier Server
* Check  using TEST Button 
* Receive PopUp Connection to Repetier works correctly
* Fill in API Key per Cut-and-Paste from RepetierServer-WebSite 
* Try to Fill empty DropDown Field Printer using Refresh Printer Button 

* Expected to get Printer(s) in the Printer DropDown List 
* Program Crash with "Speicherzugriffsfehler (Speicherabzug geschrieben)" means Segmentation Fault / core dumped

### All Original Messages on Console after Start of Slicer :

Gtk-Message: 16:26:02.363: Failed to load module "canberra-gtk-module"
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.664: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.664: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.664: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.664: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.664: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.664: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.664: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.664: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.761: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.761: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.761: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.761: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.769: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.769: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.769: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.769: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.769: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.769: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.769: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.769: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.772: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.772: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.772: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.772: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.772: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.772: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.772: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.772: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.774: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.774: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.774: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.774: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.775: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.775: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.775: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.775: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.777: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.777: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.777: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.777: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.811: invalid (NULL) pointer instance
[... 122 lines cut ...]
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.835: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.836: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.836: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:02.836: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:02.836: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.651: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.651: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.651: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.651: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.653: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.653: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.653: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.653: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.654: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.654: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.655: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.655: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.656: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.656: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(prusa-slicer:1166849): GLib-GObject-WARNING **: 16:26:13.656: invalid (NULL) pointer instance
(prusa-slicer:1166849): GLib-GObject-CRITICAL **: 16:26:13.656: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Speicherzugriffsfehler (Speicherabzug geschrieben)

 ### Screenshot 
[image]

rtyr, 2020-11-10:
Already reported here https://github.com/prusa3d/PrusaSlicer/issues/5083. Closing.

closed 2020-11-10 by rtyr

whackl, 2020-11-10:
setting the printer name in the repetierserver URL like http://A.B.C.D/#!/printer/PRINTER_NAME/ in the file ~/.config/PrusaSlicer-alpha/physical_printer/PRINTER_NAME.ini under the variable printhost_port 
did the trick ...
so the file looks like :

# generated by PrusaSlicer 2.3.0-alpha3+linux-x64 on 2020-11-10 at 16:34:53 UTC
host_type = repetier
preset_name = Original Prusa i3 MK3S MMU2S
print_host = <A.B.C.D>
printer_technology = FFF
printhost_apikey = <XXXXXXXXX>
printhost_authorization_type = key
printhost_cafile =
printhost_password =
printhost_port = Prusa_i3
printhost_user =