prusa3d/PrusaSlicer #5120: [2.3.0-alpha3] Slicing gets very slow after several slices

vintagepc, 2020-11-10:
### Version
2.3.0-alpha3

### Operating system type + version
OpenSuSE 42.3 x64

### Behavior
It seems like after several slices, the slicing process gets extremely slow - the bar sticks at around 95% for 5+ minutes - yet the only thing I changed between this and the previous slice was the bed temperature; The initial few slices of this plate when I was actively adjusting settings like perimeters were quite quick. 

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Figures.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5518403/Figures.3mf.gz)

bubnikv, 2020-11-10:
Is the amount of RAM consumed by PrusaSlicer growing with each slicing?

vintagepc, 2020-11-10:
Sorry, red herring.  Looks like the cause is actually 'Avoid Crossing perimeters' that got checked. Slicing the example gcode with that takes 10+ minutes on my machine, turning it off it is sliced in a matter of seconds.

bubnikv, 2020-11-10:
thanks. Rework of the "Avoid crossing perimeters" feature is in the queue.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

closed 2020-12-11 by bubnikv