prusa3d/PrusaSlicer #5122: Preview shows hole but not present after slicing

jcwren, 2020-11-10:
### Version
2.3.0-alpha3+win64

### Operating system type + version
Windows 7 64-bit
PrusaSlicer
Version:   2.3.0-alpha3+win64
Build:     PrusaSlicer-2.3.0-alpha3+win64-202011041552

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition
Total RAM size [MB]: 17,054MB
OpenGL installation
GL version:   4.6.0 NVIDIA 390.65
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1060 3GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 312 lines cut ...]
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa MK3 3.9.0

### Behavior
Object is manifold, renders square hole upper right in preview but does not slice it, and no errors in object are reported. 

#### Project File (.3MF) where problem occurs
[image]
[image]

[Solder.zip](https://github.com/prusa3d/PrusaSlicer/files/5520083/Solder.zip)

neophyl, 2020-11-10:
Weird, I opened your saved 3mf and sliced it and the missing square hole slices fine.  Im on the same version and same Win 64 too so I don't know what the difference is.  Try opening your saved 3mf and see if it slices.  Sometimes saving as 3mf can actually fix things (rarely).

Regardless of that the model is displaying with errors (the yellow triangle).  So try the right click and fix through netfabb option on the model which when I tried it did repair it and remove the yellow warning.

btw in case you are interested the 3dprint tool in Blender reports the following issues with the model
4884 Intersecting Faces
6240 Zero Faces
92 Zero Edges
262 Non Flat Faces
55 Thin Faces

jcwren, 2020-11-10:
OK, that's bizarre. I load the STL (which I've zipped and attached), and I get no errors on load or after slicing. Yet I load the .m3f project (as you suggested), and I see those errors.

[Solder_STL.zip](https://github.com/prusa3d/PrusaSlicer/files/5520362/Solder_STL.zip)

neophyl, 2020-11-10:
Well the model definitely has issues and seeing as the fix through netfabb option is available to you as you are running PS on Windows why not just use it to fix the model ?  

Unless you are using it as a learning experience to figure out why the model has issues (assuming its a model you created) ?  Sometimes figuring out why its not good is time well spent when it comes to making/designing/exporting other designs.

jcwren, 2020-11-11:
It's more a matter of if there are errors, why did it take saving the file to a .m3f and then opening the project to be alerted to them? If there are errors, they should be indicated when the .stl is loaded or after it's sliced. It's not a problem to fix them from within PS or with an online tool **if I know they're there**.

As far as the level of detail in this particular model, it's a trade-off. This is an STL generated from OpenSCAD. On larger circles, if I don't set the `$fn` parameter to a large value, then circles are not actually round. The trade off, of course, is that on small circles and cylinders, like those three on the right have a ridiculous level of detail. It may be possible to set the `$fn` parameter on a per circle basis, but then I don't really care. If it renders and prints and has excess detail, it's not like it's going to damage the printer or take significantly longer to print. It just takes a little longer to render in OpenSCAD and to upload to the printer. Regardless, the level of detail shouldn't be breaking anything, if it indeed that's what's causing PS to choke on rendering.

neophyl, 2020-11-11:
Its not the size that PS is choking on, and you are right it should have flagged the issues when imported in so thats a bug.

The geo comment was more a wtf comment :)  I opened it in blender to look at the actual geometry and it really slowed it down.  OpenScad does have quite a history of generating problematic models where Slicer is concerned.  Many of the problem ones people post about on the forum are from that program.  

The actual geo overlaps in places, mostly on the rounded edges on the external of the model, not so much on the circular holes.

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo

bubnikv, 2021-07-02:
Released with PrusaSlicer 2.3.2-beta.