prusa3d/PrusaSlicer #5123: Poor print quality when sliced with PrucaSlicer

f-hnt, 2020-11-10:
### Version
_Version of PrusaSlicer used: 2.2.0+win64

### Operating system type + version
Windows 10, x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s, stock version (no modifications). Firmware: 3.9.1

### Behavior
The print quality is poor with uneven surface and deep grooves when the model is sliced with PrusaSlicer. The comparison is made with GCODE generated from Simplify3D.

** _Steps needed to reproduce the problem_**

- PrusaSlicer: 
Slice the model "Model.stl" with the default printer profile.
Printer profile: "0.2mm SPEED" (tested also on "0.2mm QUALITY" with the same result)
Filament profile: "Prusa PLA"
Modifications:
Perimeters: 5
Infill: 40%

- Simplify3D: 
Slice the model "Model.stl" with the default printer profile:
Printer profile: default profile for MK3/MK3s according to the manual: 
Filament profile: "PLA"
https://help.prusa3d.com/en/guide/how-to-import-profiles-to-simplify3d-4-x-windows-macos_18360#_ga=2.99602526.1392206463.1604910174-1667211299.1604646890
Modifications:
Perimeters: 5
Infill: 40%

Both were printed with the same material, same orientation  and in the same environment without any changes in between. 
Also tested with Prusa PETG, with the same results.

Orientation: 
[image]

Results:
[image]

I'm attaching both GCODEs that were used (from PrusaSlicer and Siplify3D), the project file from PrusaSlicer and the .STL file:
[Output files.zip](https://github.com/prusa3d/PrusaSlicer/files/5520081/Output.files.zip)

rtyr, 2020-11-12:
Enable "Detect thin walls". It should help in this case.

f-hnt, 2020-11-13:
It definitely improved the quality of the print, but still Simplify3D deliver better results:
[image]

Why this option is not enabled by default and what are the drawbacks of using it all the time?

Thank you, 
Stefan

rtyr, 2020-11-13:
It shouldn't be enabled by default since it can introduce other problems in some cases, see #227, #340, #149, #3611. #4110, #2583, #4137.

closed 2020-11-13 by f-hnt