prusa3d/PrusaSlicer #5124: Apple Silicon

larsenv, 2020-11-11:
Will PrusaSlicer be updated for Apple Silicon Macs?

bubnikv, 2020-11-11:
Eventually yes.

We will see (and the rest of the world with us) how easy that will be.

tagy, 2020-11-11:
For what its worth I have been using the existing intel version on the dtk for a while without issues.

Obviously I have not tried everything, but hopefully an option until a native version.

bubnikv, 2020-11-11:
Thanks. What is the performance of the PrusaSlicer x86 build on an ARM OSX
machine?

tagy, 2020-11-12:
I did try to send you an email @bubnikv, but incase you did not get it, I think the performance is fine and perfectly usable. (plus the shipping macs are quite a bit faster)

bubnikv, 2020-11-20:
@tagy thanks, I have received the e-mail.
We already have one mac air available, we will try to compile on it.

bubnikv, 2020-11-21:
https://developer.apple.com/documentation/xcode/building_a_universal_macos_binary
https://www.starmessagesoftware.com/blog/compile-macos-desktop-application-arm64-apple-silicon-m1-chip-cpu

xarbit, 2020-11-25:
@bubnikv 
I compiled cmake and gettext on my MacBook Pro M1 .. installed to /usr/local .. 
When I try to build the dependencies after running make, I get the following error:

```
[  9%] Performing build step for 'dep_boost'
project-config.jam:25: in modules.load
*** argument error
* rule option.set ( name : value ? )
* called with: ( prefix : /Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/destdir/usr/local )
* extra argument SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/destdir/usr/local
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/util/option.jam:13:see definition of rule 'option.set' being called
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/build-system.jam:255: in load-config
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/build-system.jam:486: in load-configuration-files
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/build-system.jam:607: in load
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/kernel/modules.jam:295: in import
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/tools/build/src/kernel/bootstrap.jam:139: in boost-build
/Volumes/Extreme SSD/Development/Private/PrusaSlicer-version_2.3.0-alpha4/deps/build/dep_boost-prefix/src/dep_boost/boost-build.jam:17: in module scope
make[2]: *** [dep_boost-prefix/src/dep_boost-stamp/dep_boost-build] Error 1
make[1]: *** [CMakeFiles/dep_boost.dir/all] Error 2
make: *** [all] Error 2
```

Any idea what I am doing wrong?

bubnikv, 2020-11-25:
Sorry, we cannot help yet, as we did not get our hands on the new Apple
Silicon device yet.

bubnikv, 2020-12-01:
See https://github.com/prusa3d/PrusaSlicer/issues/5179#issuecomment-736705450
Closing

closed 2020-12-01 by bubnikv

bubnikv, 2021-04-14:
Implemented in PrusaSlicer 2.3.1-rc