prusa3d/PrusaSlicer #5128: 2.3.0 Unknown extrusion issues

Pun-e, 2020-11-11:
### Version
2.3.0-A3

### Operating system type + version
W10 64

### Behavior
No matter what combination of profiles, I'm suddenly suffering from extreme overextrusion or maybe an incredibly bad linear advance setup.

[image]
This is a purge tower, first layer. You can see how fat the straight lines are, and the corners have basically no filament

Here is the exact same tower, but sliced with 2.2.0
[image]
It's fine

[image]

The same issue can be seen in the actual object that I was trying to print a 2 colour PETG print. It was only up to the third layer and you can already see how massively over-extruded everything is and the outer perimeter has lost all definition.

This issue has persisted for multiple application restarts but I have no way of telling whether or not the gcode will be affected until I actually try to print it

#### Project File (.3MF) where problem occurs

[Mystery_Extrusion.zip](https://github.com/prusa3d/PrusaSlicer/files/5524785/Mystery_Extrusion.zip)

rtyr, 2020-11-11:
Your K value is way too high. Typical PETG value is K0.08.

[image]

Pun-e, 2020-11-11:
I just realised the same thing and I'm so embarrassed!

Thank you, waiting for my PS5 got me making all kinds of errors 🙄

closed 2020-11-11 by Pun-e