prusa3d/PrusaSlicer #5129: 2.3.0-A3 Seam problems

LeonidSl, 2020-11-11:
### Version
2.3.0-alpha3+win64

### Operating system type + version
Win 8.1 | 64

### 3D printer brand / version + firmware version
prusa i3mk3s | firmware version 3.9.1

### Behavior
The gap at the seams
In version 2.2.0 the seam is hidden in the corner But in version 2.3.0A3 the seams are blocked in sharp corners and sharp corners are much better than in 2.2.0 but there is a problem with seams
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Current version 2.2.0 below
[image]
[image]

#### Project File (.3MF) where problem occurs
<a href="https://github.com/LeonidSl/PrusaSlicer/raw/main/Leg_250_Right_1.rar">Leg_250_Right_1.rar</a>

rtyr, 2020-11-12:
I wasn't able to reproduce your results with provided 3MF.

2 different MK3S, 3 different filament brands
[image]

LeonidSl, 2020-11-14:
The seam problem occurs when using Seam painting
[image]
[image]
[image]

rtyr, 2020-11-14:
My prints were printed using your 3MF (with your seam blockers).

rtyr, 2021-09-09:
Likely related to https://github.com/prusa3d/PrusaSlicer/issues/6186.