prusa3d/PrusaSlicer #5136: Extruder 1 is e1 and should be e0

hallerdavid, 2020-11-11:
### Version
2.2.0+

### Operating system type + version
MacOS Catalina 10.15.7

### Behavior
My Problem is, that i have an IDEX printer, and when i set Extruder 1 and Extruder 2 in the PrusaSlicer, then it uses e1 and e2 but i only have e0 and e1. 

How can i fix this? Is there a chance that you change this in further versions to a logic match?

rtyr, 2020-11-12:
Please provide your 3MF project file. I am not sure what do you mean, extruder 1 should produce T0 command.

hallerdavid, 2020-11-12:
Can´t provide it at the moment. But:

When i do a single tool print, it always uses the right T1. But i want to use the left T0. When i print IDEX all works fine.

rtyr, 2020-11-12:
It is probably just about printer profile configuration, we will see when you provide your 3MF.

hallerdavid, 2020-11-12:
Here it is

[project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5529721/project.3mf.zip)

hallerdavid, 2020-11-12:
Settings: Extruder 1 (should be left) but uses the right tool.

rtyr, 2020-11-12:
This should help:

```
T[initial_tool]
M104 T[initial_tool] S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 T[initial_tool] S[first_layer_temperature] ; wait for extruder temp
G28 F5000
G29 S1
G1 Z10
M98 P"start_dualjob.g"
```

hallerdavid, 2020-11-12:
But i must do that every time i want to use the left one, right? So the best way is to set a custom profile for single print with left?

rtyr, 2020-11-12:
No, you don't. It will print with selected extruder.

hallerdavid, 2020-11-12:
ah! [] are variables set through prusaslicer? i haven´t to set them manuelly?

rtyr, 2020-11-12:
I am not sure I understand the question. List of available placeholders is here https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page).

rtyr, 2020-11-12:
> Edit: tried, but it doesn´t work. I think its overwritten after that gcode.?
Is far as I can see, it works at it should. G-code output is OK. I am not sure about your line `M98 P"start_dualjob.g"`. Anyway, it seems you just need to tune up your start g-code.

[image]

rtyr, 2020-11-12:
The issue is maybe in your `start_dualjob.g` file, but it is outside of the PrusaSlicer.

hallerdavid, 2020-11-12:
This is start_dualjob.g

`T0 
M98 P"purge0.g"

T1 
M98 P"purge1.g"`

and this is the purge code (for example for T0):

`G28 X0
T0 
M83
G1 E15 F150 
G1 E-1
G92 E0 
M82

M564 S0

G1 X35 F15000
G1 X-1
G1 X25
G1 X-1
G1 X10

M564 S1`

rtyr, 2020-11-12:
So the issue is indeed caused by start_dualjob.g, where the last tool command is T1. So if you place `T[initial_tool]` line to the end of the start gcode it may work.

hallerdavid, 2020-11-12:
Yes! Works now! Thank you! :-)

closed 2020-11-12 by hallerdavid