prusa3d/PrusaSlicer #5137: [2.3.0-alpha] Paint Seam and View Seam

amioranza, 2020-11-12:
Hi,

The paint seam feature for the alpha version is not being respected. I did some prints where I defined the seam at some insets of the print but the seam was positioned as defined at the settings, my case Rear.

Another thing is when we try to see where the seam will be positioned when viewing the sliced preview is almost impossible to identify where the seam is located. 

I'm using the Ubuntu 20.04 and the PrusaSlicer 2.3 alpha 3, not gtk3 version.

Thanks.

lukasmatena, 2020-11-12:
Please, post a 3MF. The "I did some prints" does not allow us to reproduce it. Thanks.

amioranza, 2020-11-12:
Most of the files has copyright and I can't attach it to GitHub, but this is one of the examples where the seam was not respected.
x-special/nautilus-clipboard
copy
file:///home/ghost/Documents/3d_models/tmnt-bust-fan-art/tmnt-bus_seam_paintt.3mf

bubnikv, 2020-12-05:
Your issues should be fixed in PrusaSlicer 2.3.0-beta2.

> Another thing is when we try to see where the seam will be positioned when viewing the sliced preview is almost impossible to identify where the seam is located.

That is something that we will most likely not manage to squeeze into PrusaSlicer 2.3.0 final release, but it will be most likely implemented in 2.4.

closed 2020-12-05 by bubnikv