prusa3d/PrusaSlicer #5138: Linear Advance should have a dedicated entry in filament settings

Pun-e, 2020-11-12:
### Behavior
Right now managing your linear advance is handled by simply adding M900 into the filament's respective start gcode section. This is an adequate solution for single-extruder setups, but quickly shows its limitations for any multi-material systems.

I'm integrating a TPU gasket into a PETG shell with quite tight tolerances, and I believe I'm (ironically) running into the issue of neither LA value being good enough for both of these materials. I attempted to triage my own problem with the tool change gcode slot but then quickly realised I would not be able to dynamically fetch the filaments k factor as it's not a proper variable that the slicer keeps track of.

If LA was treated like the extrusion multiplier, or temperature, it would give the slicer the ability to intelligently set the LA as needed. I also believe that better exposing LA to new users would help them to understand its importance, and hopefully reduce the calls for band-aid solutions such as coasting, wiping, etc.

_Is this a new feature request?_
Yes.

lukasmatena, 2020-11-12:
I don't understand the problem. Even if it was a separate option, you'd still need to use two separate filament presets if you wanted the values to differ for the two filaments. And you can do exactly that with the filament start gcode, can't you? You are right that custom gcode don't have access to the value now, but you should not need it in your case.

Pun-e, 2020-11-12:
PETG Start Gcode: M900 K0.1
TPU Start Gcode: M900 K0.2

When printing these two filaments _simultaneously_, only one of these k-factors will be stored and used by the printer

Does the problem make sense now?

lukasmatena, 2020-11-12:
> The only way around this would be if the slicer inserted the k-factors into the gcode whenever the tool changed

I live under impression that it does just that. If it doesn't it should. I will verify that. Thanks.

rtyr, 2020-11-12:
I think it works as it should. Filament g-code is inserted after each toolchange. @Pun-e can you provide your 3MF?

Pun-e, 2020-11-12:
I was trying to be helpful, but I really am not giving very good feedback right now 😅

I just analysed some gcode and you're totally right that the filament's start gcode is inserted with every toolchange.

It's even in the tooltip my goodness I should just go to bed

Thanks

closed 2020-11-12 by Pun-e

rtyr, 2020-11-12:
Hopefully your PS5 will arrive soon :-).