prusa3d/PrusaSlicer #5141: 3Dconnexion not work after importing stl (focus problem ?)

provalinf, 2020-11-12:
### Version
2.3.0 alpha 3 (but same alpha 2/1)

### Operating system type + version
Windows 10 x64 2004 20241.1005
3DxWare 10.7.0.3248
3DxWinCore 17.7.0.17687

### Behavior
* (Small) Problem after importing (drag and drop or import button) stl file to move the view with the 3Dconnexion joystick
* 1. Import STL
* 2. Try to move model with 3DConnexion (spacemouse for me)
  * No work
* 3. Click in the window (switch to another tab or other) and it works again. (While Prusaslicer has however well recovered the focus after the import)

### GIF for details
[image]

bubnikv, 2020-12-11:
Maybe we cannot do anything about it? As long as the driver shows "ShowX" instead of PrusaSlicer, it does not pipe the events to PrusaSlicer, but to ShowX application.

You may try to contact 3DConnexion support.

provalinf, 2020-12-15:
I wasn't clear enough, ShareX is the software that saves the screen. The problem is just after the drag and drop and it's PrusaSlicer which is indicated by the driver at this time, except that as you can see I have to click again in the window which deselects the object for the joystick to work. I can film the screen and the action on the joystick for more details if needed.

bubnikv, 2020-12-15:
I see it. It is a general focus issue. After a pop-up dialog is shown to open a file, and then closed, PrusaSlicer does not acquire keyboard and mouse focus.

bubnikv, 2020-12-15:
This is a regression, it worked in PrusaSlicer 2.2.0

bubnikv, 2020-12-15:
I think we have had this issue in PrusaSlicer 2.2.0 on Linux, where we have patched it. Now we have it on Windows as well after ugrading wxWidgets. Hopefully this will patch it on Windows.

20f5b7a7cd79c8797c6cb12c41ecf767e0258e28

provalinf, 2020-12-17:
It's all good, I confirm you that it works perfectly now with the PrusaSlicer 2.3.0-rc1 version!
Thank you

closed 2020-12-17 by provalinf